genodelabs/genode #2012: Supplement base/log.h with raw output function

nfeske, 2016-06-16:
The new `base/log.h` interface features functions for printing log messages, warnings, and errors. For low-level debugging, it would be convenient to also have a `raw` function that targets the platform-specific kernel debugger, if available. This would spare us the hassle with dealing with archaic kernel interfaces directly when working on low-level framework code.

skalk, 2016-06-17:
Maybe we can use this also within core's bootstrap code, thereby making a core-specific log implementation superfluent?

nfeske, 2016-06-17:
The core-specific log backend is still needed (because core does not open a LOG session like normal components so) but by using the new raw_write_string function, the log backend could be implemented in a kernel-agnostic way. The only problem is NOVA where the function is empty at the moment. Here the current log backend (base-nova/src/include/base/internal/core_console.h) uses a serial driver because the kernel has no syscall to output text.

skalk, 2016-06-17:
I see, thanks for the explanation.

commit 56ede24b (2016-06-16): Supplement base/log.h with raw output function

This patch introduces the Genode::raw function that prints output
directly via a low-level kernel mechanism, if available.

On base-linux, it replaces the former 'raw_write_str' function.
On base-hw, it replaces the former kernel/log.h interface.

Fixes #2012
commit 2030ae67 (2016-06-22): Supplement base/log.h with raw output function

This patch introduces the Genode::raw function that prints output
directly via a low-level kernel mechanism, if available.

On base-linux, it replaces the former 'raw_write_str' function.
On base-hw, it replaces the former kernel/log.h interface.

Fixes #2012
closed 2016-06-22 by chelmuth