genodelabs/genode #2018: Heeselicht scenario

skalk, 2016-06-21:
At this year's Hack'n'Hike, I've tried to make my daily Genode operating system setup reproducible for others. This issue documents how to obtain a USB live system featuring Genode. The setup is able to make your so far used, unmodified operating system available to you within a VirtualBox VM, while you're playing with Genode natively.

skalk, 2016-06-21:
The Heeselicht scenario is based on the Turmvilla scenario provided by nfeske. In contrast to the original scenario, you are not forced to install a specific Linux setup on a designated machine. You can use your home-computer with an already running operating system instead. The Heeselicht setup gets installed on an USB stick, and is booted from there.
Nevertheless be careful: this is still experimental work, and you should use it at your own risk!

The Genode branch with the Heeselicht scenario can be found [here](https://github.com/skalk/genode/tree/heeselicht). Its my personal topic branch, so if you want to follow it, I recommend you not to rely on it, because I may break its history from time to time when rebasing to Genode's development branch.

To build the USB stick's image, you need to install the latest Genode tool-chain either the binary version from the [website,](http://genode.org/download/tool-chain) or you build it from the sources. In addition, to produce the resulting image you need to install the [rump kernel tools.](http://genode.org/documentation/release-notes/14.05#Rump-kernel_tools). After that you can follow these steps:

- First, you need to create a build directory that targets the NOVA kernel in the 64bit variant: ``~/src/genode-labs/genode/tool/create_builddir nova_x86_64``
- Within the build directory, you need to modify the ``etc/build.conf`` file. Uncomment the following lines:
```
REPOSITORIES += $(GENODE_DIR)/repos/libports
REPOSITORIES += $(GENODE_DIR)/repos/ports
REPOSITORIES += $(GENODE_DIR)/repos/dde_linux
REPOSITORIES += $(GENODE_DIR)/repos/dde_rump
REPOSITORIES += $(GENODE_DIR)/repos/gems
REPOSITORIES += $(GENODE_DIR)/repos/dde_bsd
REPOSITORIES += $(GENODE_DIR)/repos/dde_ipxe
```
- Adapt the build environment by increasing the number of parallel jobs executed by make, according to your logical cpu count: ``echo "MAKE += -j8" >> etc/build.conf``
- By extending the RUN_OPT variable, you can tell the build environment to produce a disk image in the end: ``RUN_OPT += --include image/disk --image-disk-size 256``
- Now, you can finally start the compilation, and disk preparation process by doing: ``make run/heeselicht``
- If you have not prepared all necessary third-party software packages beforehand, the build environment will stop in the beginning, and print a corresponding warning. In that case just copy the line after the following warning: ``You can prepare respectively update them as follows:`` and execute it.
- Finally, when hopefully everything compiled and linked correctly, the run-script will ask you for your "sudo" password to execute the 'blockdev' command. This is used to retrieve the size of your first harddisk, assuming your building-machine is the same like your test-machine. If you do not want the script to automatically retrieve the harddisk size for the VirtualBox VM configuration, you can modify the following line in the ``repos/gems/run/heeselicht.run`` run-script:
```
set vmdk_blocks [ exec sudo blockdev --getsz /dev/sda]
```
like this:
```
set vmdk_blocks "500118192" # size in 512-byte sectors
```
- Finally, when everything works fine, you can find the resulting disk image in your build directory under: ``var/run/heeselicht.img`` You can copy it to your USB stick via the 'dd' utility.

When booting with USB 3.0 sticks and a 3.0 connector, it is recommended to force the BIOS to always use USB 3.0, and not to automatically switch between 2.0 and 3.0. Otherwise it might be that the USB controller driver is slower than it needs to be.

When the boot succeeds (>= Lenovo T400 or X201 recommended with Intel wireless and graphics card), you will see a setup analogue to the described [Turmvilla scenario](https://github.com/genodelabs/genode/issues/1552). To start the operating system installed on your harddisk, type: ``start linux`` in the commandline interpreter. In the current setup it is assumed that you have at least 8 GB RAM available. If you have not enough RAM, you need to lower the RAM quota donated to the guest system. Before you start your virtual machine, you can open a UNIX-like environment via: ``start noux``. Within that environment, you can modify the VirtualBox configuration under: ``/system/vm_linux.vbox:35`` (RAMSize) and the RAM quantities in Genode's VM startup configuration under: /system/subsystems/linux.subsystem`` accordingly.

skalk, 2016-06-21:
In addition, when you successfully booted Genode from USB stick and your previous installed OS from harddisk in a VM, I would recommend you to install the [VirtualBox guest additions](http://download.virtualbox.org/virtualbox/4.3.16/VBoxGuestAdditions_4.3.16.iso) inside your guest OS. Thereby you enable features like mouse-pointer integration, VM window resizing, and most interesting shared folder support to share files between Genode and your guest OS.

parfito, 2016-06-23:
@skalk  Thank you for the clarity of this nice post.
I tried the script on my two laptops, a Dell and HP Elitebook, but they all hung up during startup, but at different stages. So I decided to order the Lenovo X201 that will soon be delivered because I am very eager to start using Genode as OS virtualization support.

skalk, 2016-06-24:
@parfito Thank you for your kind words. Too bad that it does not run on your laptops. Would you mind to provide some logging output from the failing boots?

parfito, 2016-06-24:
Here is the output from the dell, from the start until i made `start linux`
[dell_output.txt](https://github.com/genodelabs/genode/files/331742/dell_output.txt)

chelmuth, 2016-06-24:
Your log file states that the ahci_drv did not find an AHCI controller. Is this expected?

ssumpf, 2016-06-24:
@parfito: As far as I can see it, the AHCI driver does not come up
`[init -> ahci_drv] No AHCI controller found`
I assume the Linux is booted from hard  disc, which in turn will not work. Is it possible to set the hard disc controller to 'AHCI' in your BIOS?

parfito, 2016-06-24:
Ok, you are right. I just set the controller to AHCI and it is now detected. However other errors appeared such as 

> [ 0] sys_sc_ctrl: Bad SC CAP (0x2916)
> long long unsigned int Genode::Platform_thread::execution_time() const: sc_ctrl failed res=4 init -> rump_fs ->  block_io
> .....
> [ 0] sys_sc_ctrl: Bad SC CAP (0x46ee)
> long long unsigned int Genode::Platform_thread::execution_time() const: sc_ctrl failed res=4 init -> cli_monitor -> linux -> vbox ->  vbox

And finally, when I issued the `start linux `command, it prints
> [init -> ahci_drv] Device 0 not available

and the window disappears.
Here is the complete output
[dell_output6.txt](https://github.com/genodelabs/genode/files/331858/dell_output6.txt)

ssumpf, 2016-06-24:
@parfito: Okay, your hard disc is on port 1 not 0,
`[init -> ahci_drv] 		#1: ATA`
I guess you have to change the VirtualBox configuration in the run script and maybe the .vdmk file, I will have to look this up though.

ssumpf, 2016-06-24:
@parfito: Maybe it is already sufficient to set the policy of the 'ahci_drv' in heeselicht.run from
`<policy label="cli_monitor -> linux -> vbox -> sda" device="0"/>`
to
`<policy label="cli_monitor -> linux -> vbox -> sda" device="1"/>`

ssumpf, 2016-06-24:
@parfito: Also if your Linux is not on 'sda' in your build environment, you would have to change this stuff in the run script accordingly:
`puts "sudo blockdev --getsz /dev/sda"
set vmdk_blocks [ exec sudo blockdev --getsz /dev/sda]
set vmdk_fd [open [run_dir]/sda.vmdk w]
puts $vmdk_fd {# Disk DescriptorFile
version=1
CID=c976fef0
parentCID=ffffffff
createType="fullDevice"
Extent description}
puts $vmdk_fd "RW $vmdk_blocks FLAT \"/dev/sda\" 0"`

parfito, 2016-06-24:
@ssumpf, Great, It works.  I thought my machine was not yet supported. 
May be It would be possible to make it work on my HP too.... I will see it later. However, I include its output if someone want to take a look.  
[output_hp.txt](https://github.com/genodelabs/genode/files/332208/output_hp.txt)
Thank you very much for your help.

ssumpf, 2016-06-24:
:+1: Glad I could help!

skalk, 2016-06-30:
@parfito regarding your HP laptop, there is a function missing (AcpiOsStall) in our apcica library port, which is not needed when executing on top of our hardware, but on the HP it is used. I've written a small patch that should resolve that issue, can you give it a try? It's here: https://github.com/skalk/genode/commit/9bad027414f1eae6badf36733c1b56397fcfbcea

parfito, 2016-06-30:
Hello @skalk I have just tried it. Now it calls for another function - Timer. Here is the output log.
[output_hp1.txt](https://github.com/genodelabs/genode/files/341734/output_hp1.txt)

skalk, 2016-07-01:
Ah ok, seems like the acpica component did not used a timer session on my PC at all. So there is a missing route to the timer service for this component. You have to add this explicitly inside the route rules of the acpica configuration. Have a look for it in commit cc00888e53719dd8a20ad8e5374b295c5f4b62e6.

parfito, 2016-07-02:
Timer route configured, but there is still a problem.  However we are progressing. Here is the output
[output_hp2.txt](https://github.com/genodelabs/genode/files/344597/output_hp2.txt)

alex-ab, 2016-07-04:
Please cherry-pick the commits b4ffefd561f1d4c6c449fde3f8020c7f6729ab6b and 283d6bc00542cd506a17c36453b89af9ba1b0480 and post your ouput

parfito, 2016-07-04:
[Uploading output_hp3.txt…]()

parfito, 2016-07-04:
[output_hp3.txt](https://github.com/genodelabs/genode/files/346869/output_hp3.txt)

alex-ab, 2016-07-05:
According to you output the usb_drv does not manage to announce the "Block" session, therefore rump_fs will not start waiting for this announcement. @ssumpf: does the output of usb_drv tells you something suspicious ? @parfito: You may also try to add the bios_handoff option to your usb driver
``` <config uhci="yes" ehci="yes" xhci="yes" bios_handoff="no"> ```
but this is just guessing now in the dark.

ssumpf, 2016-07-05:
@parfito, @alex-ab: Actually not, there are two EHCI controllers present, two three port root hubs, and one 8 port hub. But there are no other devices (like the USB stick) that register at one of those hubs. Could you try a different USB stick?

parfito, 2016-07-05:
I tried another USB stick, then the bios_handoff option without any subsequent change or improvement compared to previous run. Here are the outputs, 
[output_hp4.txt](https://github.com/genodelabs/genode/files/348107/output_hp4.txt) for 8GB usb stick; 
[output_hp5.txt](https://github.com/genodelabs/genode/files/348111/output_hp5.txt) for turning off the bios_handoff

chelmuth, 2016-07-18:
The final life sign from usb_drv in all log files is `dev_info: new high-speed USB device number 2 using ehci-pci`. Any ideas?

parfito, 2016-07-26:
@chelmuth, what is wrong with this log?
I can even see a quite similar line (last from usb_drv) in other log files where the scenario runs correctly.

chelmuth, 2016-07-26:
The intention behind my comment was to revive the discussion. I personally have no further ideas.

skalk, 2016-10-11:
I've upgraded Genode/Heeselicht to release 16.08.

doofin, 2016-11-19:
I have followed the instructions including installing the rumpkernels , but get this strange error: 

>  make[1]: Leaving directory '/media/1t/Downloads/genode-heeselicht/build.vbox'
> genode build completed
> sudo blockdev --getsz /dev/sda
> [sudo] password for dubt: 
> cp: cannot stat 'bin/libiconv.lib.so': No such file or directory
>     while executing
> "exec cp bin/$binary [run_dir]/genode"
>     (procedure "copy_and_strip_genode_binaries_to_run_dir" line 4)
>     invoked from within
> "copy_and_strip_genode_binaries_to_run_dir $binaries"
>     (file "/media/1t/Downloads/genode-heeselicht/repos/gems/run/heeselicht.run" line 1052)
>     invoked from within
> "source $include_name"
>     ("foreach" body line 6)
>     invoked from within
> "foreach include_name [get_cmd_arg --include ""] {
> 	# first check if the include name is absolute
> 	if {[string first "/" $include_name] == 0} {
> 		puts ..."
>     (file "/media/1t/Downloads/genode-heeselicht/tool/run/run" line 642)
> Makefile:293: recipe for target 'run/heeselicht' failed
> make: *** [run/heeselicht] Error 1

Seems that libiconv is not compiled. but after i run prepare_port libiconv ,it is still missing

nfeske, 2016-11-21:
I guess that there is another prerequisite for compiling libiconv missing. Sometimes, those hints go unnoticed because they appear very early in the build log. Could you please try building only the single library via the following command and post the complete build output?

```
make LIB=libiconv
```

skalk, 2017-11-29:
This issue is outdated and replaced by the Sculpt system scenario.

closed 2017-11-29 by skalk