genodelabs/genode #2022: fs_rom does not clear dataspace on update

MartijnVerschoor, 2016-06-24:
In the _update_dataspace() function, a new dataspace is allocated when the updated file has outgrown the existing dataspace. If the file's size doesn't change or becomes smaller, the existing dataspace is reused, but it is not cleared. In the latter case, residual information from the previous file is visible.

As a crude workaround we enforce allocating a new dataspace on each update (see patch). Perhaps a better fix would be to clear the existing dataspace, but I would like to leave this to the core dev team.

[fs_rom_patch.txt](https://github.com/genodelabs/genode/files/331759/fs_rom_patch.txt)

nfeske, 2016-07-07:
Thank you for reporting! I think that I will take this issue as an opportunity to create a test for fs_rom.

chelmuth, 2017-09-29:
I guess this issue got lost. I tried to port the patch to current staging.
```diff
--- a/repos/os/src/server/fs_rom/main.cc
+++ b/repos/os/src/server/fs_rom/main.cc
@@ -225,7 +225,7 @@ class Fs_rom::Rom_session_component :
 				File_system::file_size_t const new_file_size =
 					_fs.status(file_handle).size;
 
-				if (_file_ds.size() && (new_file_size > _file_size)) {
+				if (_file_ds.size()) {
 					/* mark as invalid */
 					_file_ds.realloc(&_env.ram(), 0);
 					_file_size = 0;
```

ehmry, 2018-06-15:
This was fixed in 18.05, if not earlier.