genodelabs/genode #2023: File_system client interface lacks functions to receive signals when server acknowledges packets or is ready for more packets

NobodyIII, 2016-06-26:
In my fs_filter component (much like the vfs server), I am trying to allow multiple clients fast, non-blocking access to multiple filesystems. To do this without several threads, my component needs to know when each filesystem server unblocks so it can send more packets. Unfortunately, the File_system::Client interface (which File_system::Connection inherits) doesn't provide a way to receive signals when the server acknowledges packets or is ready for more packets, given that it only provides a Session:Tx::Source object for accessing the packet stream. The Session::Tx::Source interface provide the methods sigh_ack_avail() and sigh_ready_to_submit(), which allow a cilent to get the signal handlers currently in use, but doesn't allow registering new signal handlers. I added a workaround that exposes the Packet_stream_tx::Client object to the client, but it would be good to have a better way to register these signal handlers. Is there currently a way to do this, or does this require a modification to the API.

ehmry, 2016-07-12:
This is a requirement for an asynchronous File_system VFS plugin, the client needs a non-blocking way of knowing when acks have been queued.

nfeske, 2016-07-12:
The commit https://github.com/NobodyIII/genode/commit/db21ae2575e3c68d4955865d3d40b999d21f2fdf fixes the issue. I am waiting for the GCA signed by @NobodyIII before I can go forward with merging it into genodelabs/staging.

ehmry, 2016-07-12:
Excellent

nfeske, 2016-07-13:
I added the patch https://github.com/genodelabs/genode/commit/31cccf41af3f142e0fc8268dd47daaa7cb34b4ad to staging. @NobodyIII I refined the commit message (adjusted the subject to fit 50 characters, added a reference to the issue, and replaced your nick by your real name). Are you fine with those changes?

NobodyIII, 2016-07-13:
> I added the patch 31cccf4

Sure. Sounds good.

commit 31cccf41 (2016-07-13): os: let fs clients install custom signal handlers

This patch adds the methods 'sigh_ack_avail()' and
'sigh_ready_to_submit()', which are needed to build asynchronously
operating file-system clients.

Fixes #2023
commit a4f73ab1 (2016-07-15): os: let fs clients install custom signal handlers

This patch adds the methods 'sigh_ack_avail()' and
'sigh_ready_to_submit()', which are needed to build asynchronously
operating file-system clients.

Fixes #2023
closed 2016-07-15 by nfeske