genodelabs/genode #2030: base-linux: nic_drv causes segmentation fault in libc

ValiValpas, 2016-07-02:
I'm encountering segmentation faults when I use the nic_drv on base-linux. I've already noticed this some time ago but I couldn't find a corresponding issue here on github.

When I run e.g. `netperf_lwip` with the 16.05 release, I get the following output:
`[init -> nic_drv] Segmentation fault (signum=11), see Linux kernel log for details`

dmesg shows some more details:
` nic_drv[7628]: segfault at 2201c18 ip 00007f434df02b83 sp 00007fffa75609d0 error 4 in libc-2.23.so[7f434de8a000+197000]`

cproc, 2016-07-05:
One recently introduced bug is that currently a stack area gets attached in hybrid Genode/Linux applications (like the nic_drv). If I remember correctly, it was at this code location: https://github.com/genodelabs/genode/blob/93845e1de5085395da6c3b75630aab129b71e722/repos/base-linux/src/lib/base/platform_env.cc#L152. This can have the effect that a memory region which is already in use (for example by the libc) can be overmapped with this stack area.

nfeske, 2016-07-06:
@ValiValpas could you please give commit 12be17d475091e05543494ce1ea28e8eb2e7da29 a try?

ValiValpas, 2016-07-07:
@nfeske I'd love to, however, I struggle fetching the specific commit (it's not an a branch).

nfeske, 2016-07-07:
The corresponding branch is here: https://github.com/nfeske/genode/commits/lx_hybrid_stack_area

ValiValpas, 2016-07-07:
Thanks. The commit does indeed fix this issue.

nfeske, 2016-07-07:
Cool, thanks for testing! :-)

commit 6cae7ea9 (2016-07-07): base-linux: omit stack-area init for lx_hybrid

Fixes #2030
commit 57ec61fb (2016-07-11): base-linux: omit stack-area init for lx_hybrid

Fixes #2030
closed 2016-07-12 by nfeske