genodelabs/genode #2032: ahci: overlapping request may lock up the device

cnuke, 2016-07-05:
```
[…]
[init -> test-block_replay] start replaying 22145 requests bulk=yes tx_size=2097152
init -> test-block_replay] Warning: could not submit request
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> test-block_replay] Warning: could not submit request
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> test-block_replay] Warning: could not submit request
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 0 + 1, request: 0 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51881 + 1, request: 51881 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51881 + 1, request: 51881 + 1
[init -> ahci_drv] Overlap: pending 51890 + 1, request: 51890 + 1
[init -> ahci_drv] Overlap: pending 51891 + 1, request: 51891 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51881 + 1, request: 51881 + 1
[init -> ahci_drv] Overlap: pending 51890 + 1, request: 51890 + 1
[init -> ahci_drv] Overlap: pending 51891 + 1, request: 51891 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51880 + 1, request: 51880 + 1
[init -> ahci_drv] Overlap: pending 51881 + 1, request: 51881 + 1
[…]
```

nfeske, 2020-02-28:
@ssumpf is this issue still valid after your great refactoring of the AHCI driver?

ssumpf, 2020-02-28:
@nfeske: Maybe, the check has not changed much, it would be good to run the test again I think.

cnuke, 2020-02-28:
I am not sure if I still have the `block_tester` config, which triggered the problem. I will look into it.

cnuke, 2020-03-04:
I dug up the old branch containing d0e1c84, which includes the sequence of requests triggering the problem AFAIR. I will try if I can reproduce it when replaying the request sequence once more using the `block_tester` component with staging but I will nit spend time rebasing the original commits. After all we have not encountered any practical overlapping issues so far.

cnuke, 2020-03-04:
I tested d08b355 on hardware where indeed the overlay check is triggered (as expected) but the AHCI device does not lock up.  There I will close the issue.

closed 2020-03-04 by cnuke