genodelabs/genode #2034: serve next layer configuration from Nic session

ehmry, 2016-07-06:
Nic servers that provide interfaces to overlay networks seem to have usually IP addressing information present, for example, OpenVPN, cjdns, and ZeroTier all assign their own IP addresses. TOR (#2026) also has something like this with the **TransPort** and **VirtualAddrNetwork**. 

Perhaps the Nic session could be amended with a method like ``Rom_dataspace_capability config()``. In the default case, such as with Nic_bridge, the server would return an invalid dataspace. For servers that provide addressed interfaces a dataspace with XML configuration could be returned in a form like:

```
<config>
  <ip>
    <v4 addr="172.16.254.2" netmask="255.255.255.0" gateway="172.16.254.1"/>
  </ip>
  <mtu bytes="1500"/>
</config>
```

I'm not sure how much work it would be to modify the lwip and lx stacks to use this convention, but this may re-order some of the addressing configuration.

ehmry, 2016-07-06:
Additionally, if the configuration was published to a report or served as ROM, it could be then attached to a VirtualBox shared folder thru the VFS, and then a script native to the VM OS could setup addressing on the interface.

This line of thought started with embedding a DHCP server at a Nic session, but this seems safer to implement.

nfeske, 2016-07-06:
Interesting idea. We could even take it a bit further by embedding an actual ROM session into the NIC session (similar how nitpicker aggregates a framebuffer and input session). This way, the ROM could even carry dynamic information about the network such as the link state. Also the MAC address could be supplied to the client in this way.

ehmry, 2016-07-26:
I have a long time interest in overlay networks, so I went ahead and implemented a good deal of this.

Right now I am able to run a [ZeroTier](https://github.com/zerotier/ZeroTierOne) Nic server, connect that to VirtualBox, and export the interface configuration to a guest OS using the VFS. In this scenario, the guest operating system receives a virtual MAC address and IP address, but the addressing key-pair and physical IP stack is outside the virtual machine in a native component.

I have the ZeroTier port ready for the world repository if this clears. I met the author @adamierymenko in Berlin this spring and he was able to explain to me the porting process in just a few minutes. I believe same method could be applied to OpenVPN or TOR (overlooking the lack of UDP) if virtual Nic servers simply include addressing in this ROM sub-session.

Commits 74bacffa93fa88ad54b16b60ac50f026e9a1670a and 58fe00931057091a2db011923bf508b0fab5d873 are boring and only deal with component API changes.

ceeeba798a640468a3dd11881520790606d8d45d implements the embedded ROM session and is backwards compatible with all Nic clients.

e98fc4d5a1b8f6c7dd093663d84212dca09d208b adds IPv4 addressing information to the embedded ROM session and adds support for nic_bridge.

ec41b203d63dc0ebeae9653b6375586273b8e9e4 introduces a server for reporting this ROM session thru an independent component, a Report client could have been embedded into the generic Nic component base class but I thought best to keep it lean for now.

3479b7df3b46ed3ae7daf3ca4c76bb61ccc3fa9f adds IPv4 address parsing to lwIP, lwIP will use addresses from the Nic ROM session if no static IP addresses are defined in the the config.

LxIP support is missing.

This is the configuration I mentioned earlier, in this example the VMM has two interface, 0 is routed to parent, 1 is routed to the overlay using port labeling from #1966.

```xml
	<start name="zerotier">
		<exit propagate="yes"/>
		<resource name="RAM" quantum="16M"/>
		<provides><service name="Nic"/></provides>
		<config>
			<vfs> <fs/> </vfs>
			<default-policy name="linux" network="565799d8f6cbdca2"/>
		</config>
	</start>

	<start name="report_rom">
		<exit propagate="yes"/>
		<resource name="RAM" quantum="1M"/>
		<provides>
			<service name="Report"/>
			<service name="ROM"/>
		</provides>
		<config verbose="yes">
			<policy label="vbox -> nic.xml" report="nic_report -> vbox -> 1"/>
		</config>
	</start>

	<start name="nic_report">
		<exit propagate="yes"/>
		<resource name="RAM" quantum="1M"/>
		<provides><service name="Nic"/></provides>
		<route>
			<service name="Nic">
				<child name="zerotier"/> </service>
			<service name="Report">
				<child name="report_rom"/> </service>
			<any-service> <parent/> </any-service>
		</route>
	</start>

	<start name="vbox" priority="-1">
		<exit propagate="yes"/>
		<binary name="virtualbox" />
		<resource name="RAM" quantum="800M"/>
			<config vbox_file="/vbox/vm_linux.vbox" vm_name="linux">
			<libc stdout="/dev/log" stderr="/dev/log" rtc="/dev/rtc">
				<vfs>
					<dir name="dev">
						<log/>
						<rtc/>
						<block name="sda" label="sda" block_buffer_count="128" />
					 </dir>
					<dir name="vbox">
						<fs label="usb"/>
					</dir>
					<dir name="shared">
						<rom name="nic.xml"/>
						<fs label="shared"/>
					</dir>
				</vfs>
			</libc>
		</config>
		<route>
			<service name="Nic" label_suffix="1">
				<child name="nic_report"/> </service>
			<service name="ROM" label="nic.xml">
				<child name="report_rom"/> </service>
			<any-service> <parent/> <any-child/> </any-service>
		</route>
	</start>
```

m-stein, 2016-08-09:
@ehmry I really like your ROM-config approach. I plan to draw on it in the NAT issue (#114) and the DHCP component I like to write a  from scratch as soon as the NAT is finished.

ehmry, 2016-08-09:
Ok good, I'll plan on patching lxip for pulling in static information too.

ehmry, 2018-06-04:
Closing for now.

closed 2018-06-04 by ehmry