genodelabs/genode #2040: asynchronous VFS I/O

ehmry, 2016-07-12:
@chelmuth's work to expose a common network stack through a shared file system requires the VFS server to perform network I/O on behalf of a client without blocking other clients. I have an initial implementation of the callback API that has been discussed, but so far it has only been run against the current test scenarios. It has not been benchmarked, as comparisons cannot be made until we have scenarios that utilize the callbacks rather than operate through wrappers.

A quick breakdown of the changes:

Reads and writes pass from the VFS to the application thru callbacks. The application sets callbacks on a handle and then uses a ``read`` or ``write`` call to schedule the handle callbacks. These calls may return ``READ_QUEUED`` or ``WRITE_QUEUED`` in addition to the existing I/O errors.

The callback is passed a buffer from which to read and write from, and the status of the callback. If the callback receiver should expect additional callbacks, the status will be ``Callback::PARTIAL``, if the callback receiver will have completed the read or write at the end of the current call, ``Callback::COMPLETE`` is passed. If the stream of callbacks is terminating due to an error, ``Callback::ERROR`` is passed.

If the callback has been executed before the ``read`` or ``write`` call returns, ``READ_OK`` or ``WRITE_OK`` indicates the operation completed without queuing. Wrappers have been provided to emulate the current synchronous API by waiting for inline callbacks to complete before returning from read and writes.

The VFS server has been adopted to use callbacks. Now every internal file node that corresponds to a client file handle contains a packet descriptor. One packet operation may be queued at each file handle, and these packets are dequeued by callback or when the file is closed. Scheduling additional packets before the callback has completed will forward the incomplete packet to the client. This increases the number of in-flight packets to the number of open file handles, but the client is still constrained by the size of the bulk buffer when queuing operations.

Again, this is not that well tested, and definitely needs further discussion.

nfeske, 2016-07-12:
Impressive work!

ehmry, 2016-07-18:
I have an async version of the terminal plugin, put I don't think its useful until libc task switching is implemented.

nfeske, 2017-02-28:
Fixed in Genode 17.02

nfeske, 2017-02-28:
@ehmry I see several related commits in this issue, like the highly anticipated pipe-VFS-plugin. Let us move those features to dedicated issues now that the basic foundation is in place.

closed 2017-02-28 by nfeske

ehmry, 2017-02-28:
Agreed.