genodelabs/genode #2046: wifi: scheduled scanning broken on Intel MVM based cards

chelmuth, 2016-07-21:
On MVM only the initial scanning of APs works. If wpa_supplicant changes from triggered scans to the scheduled-scanning hardware feature no further scan results are reported. Relevant log snippets follow.
```
[init -> wifi_drv] dev_info: loaded firmware version 16.242414.0 op_mode iwlmvm
[init -> wifi_drv] dev_info: Detected Intel(R) Dual Band Wireless AC 8260, REV=0x208
...
[init -> wifi_drv] wlan0: Starting delayed sched scan
[init -> wifi_drv] wlan0: Use normal scan instead of sched_scan for initial scans (normal_scans=0)
[init -> wifi_drv] wlan0: Setting scan request: 0.000000 sec
...
[init -> wifi_drv] nl80211: Drv Event 33 (NL80211_CMD_TRIGGER_SCAN) received for wlan0
[init -> wifi_drv] wlan0: nl80211: Scan trigger
...
[init -> wifi_drv] nl80211: Drv Event 34 (NL80211_CMD_NEW_SCAN_RESULTS) received for wlan0
[init -> wifi_drv] wlan0: nl80211: New scan results available
...
[init -> wifi_drv] nl80211: Sched scan requested (ret=0) - scan interval 10000 msec
[init -> wifi_drv] nl80211: Drv Event 75 (NL80211_CMD_START_SCHED_SCAN) received for wlan0
[init -> wifi_drv] wlan0: nl80211: Sched scan started
<stuck>
```

chelmuth, 2016-07-21:
19c16d9 works around this issue by disabling the report of the WIPHY_FLAG_SUPPORTS_SCHED_SCAN in the MVM MAC implementation. Now, MVM is used (like DVM) with triggered scans by wpa_supplicant.

nfeske, 2018-09-03:
This issue is outdated since the big update of the wifi driver in Genode 18.08.

closed 2018-09-03 by nfeske