genodelabs/genode #2047: Avoid absolute paths to host utilities in build system

ehmry, 2016-07-27:
I don't have a /usr directory on my machine, which causes problems, but is actually not very hard to work around. I usually cherry-pick a patch onto each of my branches so this patch would make my life a bit easier :-)

chelmuth, 2016-07-27:
I wonder if (as a better alternative to your patch) the shebang lines should be fixed by using `#!env <tool>`. Admittedly, I've no experience with the portability of this approach.

cnuke, 2016-07-27:
Using `env` works in case there is no follow argument to the invoked program since you can only have one argument an in the shebang line.

ehmry, 2016-07-27:
Which would not work in the case of 'make -f'.

chelmuth, 2016-07-27:
That's a pity.

chelmuth, 2016-08-15:
Rediscovered this just now. I'm not convinced to merge the patch. IMO `/usr/bin` is a UNIX'ism but always there, and, many tools depend on it.

ehmry, 2016-08-15:
Yea, I can live with that assumption, I'm switching to Arch.

closed 2016-08-15 by ehmry