genodelabs/genode #2052: Integer wrap in Packet_stream_(sink|source)::packet_valid

ollseg, 2016-08-04:
In os/include/os/packet_stream.h the packet_valid methods add two untrusted inputs together without overflow checks. The "size" value could potentially be very large, wrapping the result of offset + size to an unexpectedly small value.

chelmuth, 2016-08-04:
Do you have a suggestion to solve the issue? From the top of my head I'd check `packet.size() <= _bulk_buffer_size` in the first line of the function.

ollseg, 2016-08-04:
That would still potentially place the end of packet outside the bulk buffer. Perhaps changing the  offending comparison to (packet.size() < (_bulk_buffer_offset + _bulk_buffer_size) - packet.offset()) to check that the available buffer space is larger than the packet size?

chelmuth, 2019-03-14:
The code of `packet_valid()` changed repeatedly since this issue was created.

closed 2019-03-14 by chelmuth