genodelabs/genode #2056: move partition enforcement from part_blk to Block::Session_component

ehmry, 2016-08-11:
I wanted to get a bit more performance of my VM's block device so this seemed like the lowest hanging fruit. In short, Block session requests now carry _offset_ and _span_ arguments that constrain the session to a region of blocks. So long as part_blk can intercept and modify these session arguments during session construction, clients are still restricted to partitions chosen by part_blk, and the intermediary packet buffer is eliminated.

I have not properly benchmarked this, but some informal tests in VBox (while trying to avoid as many caches as possible) would seem to show a significant speedup.

chelmuth, 2016-08-11:
Thanks for going forward with this idea - I much appreciate your effort. On the other hand, I think that your proposal and the current state span a spectrum of solutions and I'm not sure that we should trade one extremum for the other. I'll try to explain my thoughts in the following.

Currently, part_blk is a plain *resource multiplexer* that provides multiple clients controlled access to regions of a block device. For this purpose, it sanitizes the block-request parameters (by providing virtual block devices) and also copies block-request payload from the client to the block driver (and vice versa). The *device driver* provides only one session (which could therefore be static). The device driver is naturally much more complex than the resource multiplexer and should be completely relieved from the burden to sanitize requests from multiple clients.

I agree that we should fathom opportunities for optimization but suggest to do it step-by-step. First, we should have a base line of *numbers* to measure how much any change gains us. Second, we should identify the bottle neck, which IMO is (regarding throughput) the additional payload copy in part_blk between client and driver. If this is the case, we could try to eliminate this copy operation only, which could be easily prototyped if just benchmarking throughput with no real data.

My initial idea was to keep part_blk as hop component but support splitting the dataspace shared with the block device driver among several clients. This permits the clients to allocate request buffers in memory accessible by the driver but keeps part_blk in charge to sanitize the request parameters. This approach does not eliminate the multiple packet queues (one for each client session and one for the driver session) and the signal hop, but these mostly affect the latency not the throughput.

ehmry, 2016-08-11:
At first I realized my methodology for measuring my performance was wrong, and then today I switched the hardware I was using and got an extreme improvement . So really my perceived bottleneck was a hardware quirk, and now I don't have any issue with the block stack. It does go to show that without any baselines to compare against optimization is just stumbling in the dark.

I can understand splitting the bulk buffer between clients, and on second thought, maintaining a separate resource multiplexer is good because if fair-queuing were adding in the future, it would not make sense to me not to do that at the driver.

closed 2016-08-11 by ehmry