genodelabs/genode #2059: add virtualbox 5 support

alex-ab, 2016-08-16:


alex-ab, 2016-08-19:
@nfeske: please use the last 3 commits of [issue_2059_staging](https://github.com/alex-ab/genode/tree/issue_2059_staging)

nfeske, 2016-08-19:
Merged to staging just now. :-)

cnuke, 2016-08-22:
@chelmuth or @nfeske please merge 12903b7 and 4467f0a to staging.

nfeske, 2016-08-22:
@cnuke Thank you! The include guard of needed a small additional adjustment though ;-)

cnuke, 2016-08-22:
Shame on me — it was the first thing I thought about and still missed it at the end :-)

chelmuth, 2016-08-30:
Merged with 57b90e8.

ehmry, 2016-08-30:
I'm having an issue with network slot labeling, there is a bit of patch missing from the previous version that I can foward...

chelmuth, 2016-08-30:
Put in #2081 for post-release merge.

alex-ab, 2016-11-08:
@chelmuth: please consider d72d38d40a05cd762d16e901259e4906a26a539d for staging if it fits

alex-ab, 2016-11-08:
@chelmuth: sorry, forget about it - not ready yet because it breaks vbox4

alex-ab, 2016-11-08:
@chelmuth: commit 85347a557916d71948ec82d1c7c27b340c5f5a44 is better

alex-ab, 2016-12-01:
5.1.10 is on master

closed 2016-12-01 by alex-ab