genodelabs/genode #2060: fs_log: collect acknowledgement packets and increase stack size

cproc, 2016-08-17:


cproc, 2016-08-17:
@ehmry: I added a test case (using 'fs_log.run') with debug messages in 870b811511cde881df98819ccd15e9b1859a6c2e

It is supposed to print 30 lines of the string "text", but only 17 appear on the console.

```
Genode 16.05-200-g870b811
int main(): --- create local services ---
int main(): --- start init ---
int main(): transferred 17592186044415 MB to init
int main(): --- init created, waiting for exit condition ---
[init -> vfs] session opened for 'fs_log' at '/'
[init -> vfs] could not allocate new dataspace of size 32768
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> vfs] int main(int, char**): text
[init -> fs_log] submitting packet
[init -> vfs] not ready to ack
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] submitting packet
[init -> fs_log] packet allocation failed, getting acked packet
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> fs_log] packet allocation failed, getting acked packet
[init -> vfs] not ready to ack
[init -> fs_log] submitting packet
[init -> vfs] int main(int, char**): text
[init -> vfs] not ready to ack
```

ehmry, 2016-08-18:
The fs_log on master and staging is a version that I'd like to disown, it has some other problems.

The blocked acked problem is solved if you apply a patch to os/path.h e01744999d7e38b4250c0eaceed7abc9302e67c1 and to fs_log 1b976a9692280556b5615e4c1d5b376be30d4dbc from @chelmuth's vfs branch. If the second patch will not apply cleanly, just checkout the version of fs_log from that commit.

Then I have a fixup commit 23324c7102e4dca3abf225036d593e589670cf56 that will solve the congested ack queue problem.

closed 2016-08-30 by cproc