genodelabs/genode #2062: Session_label incompatible with Xml_node::attribute_value()

chelmuth, 2016-08-18:
The following patch tries to read a Session_label from an XML attribute but produces the attached compile error.
```diff
diff --git a/repos/base/src/test/printf/main.cc b/repos/base/src/test/printf/main.cc
index 51b617d..721afa2 100644
--- a/repos/base/src/test/printf/main.cc
+++ b/repos/base/src/test/printf/main.cc
@@ -14,8 +14,34 @@
 
 #include <base/printf.h>
 
+#include <base/log.h>
+#include <util/xml_node.h>
+#include <base/session_label.h>
+
+static char const *xml_content = "<node label=\"value\"/>";
+
+static void test_session_label()
+{
+       using namespace Genode;
+
+       Xml_node xml(xml_content);
+
+       String<32>    string = xml.attribute_value("label", String<32>("no-label"));
+#if 1
+       Session_label label  = xml.attribute_value("label", Session_label("no-label"));
+#else
+       Session_label label { "huhu" };
+#endif
+
+       log("string='", string, "'");
+       log("label='", label, "'");
+}
+
+
 int main(int argc, char **argv)
 {
+       test_session_label();
+
        /* test that unsupported commands don't crash the printf parser */
        Genode::printf("%#x %s\n", 0x38, "test 1");
        Genode::printf("%#lx %s\n", 0x38L, "test 2");
```
```
  Program test/printf/test-printf
    COMPILE  main.o
In file included from .../repos/base/src/test/printf/main.cc:18:0:
.../repos/base/include/util/xml_node.h: In instantiation of ‘bool Genode::Xml_attribute::value(T*) const [with T = Genode::Session_label]’:
.../repos/base/include/util/xml_node.h:888:10:   required from ‘T Genode::Xml_node::attribute_value(const char*, T) const [with T = Genode::Session_label]’
.../repos/base/src/test/printf/main.cc:31:79:   required from here
.../repos/base/include/util/xml_node.h:163:44: error: no matching function for call to ‘ascii_to(char*, Genode::Session_label&)’
    return ascii_to(_value.start() + 1, *out) == _value.len() - 2;
                                            ^
.../repos/base/include/util/xml_node.h:163:44: note: candidates are:
In file included from .../repos/base/include/util/token.h:17:0,
                 from .../repos/base/include/util/xml_node.h:17,
                 from .../repos/base/src/test/printf/main.cc:18:
.../repos/base/include/util/string.h:318:16: note: Genode::size_t Genode::ascii_to(const char*, bool&)
  inline size_t ascii_to(char const *s, bool &result)
                ^
.../repos/base/include/util/string.h:318:16: note:   no known conversion for argument 2 from ‘Genode::Session_label’ to ‘bool&’
.../repos/base/include/util/string.h:336:16: note: Genode::size_t Genode::ascii_to(const char*, long unsigned int&)
  inline size_t ascii_to(const char *s, unsigned long &result)
                ^
.../repos/base/include/util/string.h:336:16: note:   no known conversion for argument 2 from ‘Genode::Session_label’ to ‘long unsigned int&’
.../repos/base/include/util/string.h:347:16: note: Genode::size_t Genode::ascii_to(const char*, long long unsigned int&)
  inline size_t ascii_to(const char *s, unsigned long long &result)
                ^
.../repos/base/include/util/string.h:347:16: note:   no known conversion for argument 2 from ‘Genode::Session_label’ to ‘long long unsigned int&’
.../repos/base/include/util/string.h:359:16: note: Genode::size_t Genode::ascii_to(const char*, unsigned int&)
  inline size_t ascii_to(const char *s, unsigned int &result)
                ^
.../repos/base/include/util/string.h:359:16: note:   no known conversion for argument 2 from ‘Genode::Session_label’ to ‘unsigned int&’
.../repos/base/include/util/string.h:370:16: note: Genode::size_t Genode::ascii_to(const char*, long int&)
  inline size_t ascii_to(const char *s, long &result)
                ^
.../repos/base/include/util/string.h:370:16: note:   no known conversion for argument 2 from ‘Genode::Session_label’ to ‘long int&’
.../repos/base/include/util/string.h:399:16: note: Genode::size_t Genode::ascii_to(const char*, Genode::Number_of_bytes&)
  inline size_t ascii_to(const char *s, Number_of_bytes &result)
                ^
.../repos/base/include/util/string.h:399:16: note:   no known conversion for argument 2 from ‘Genode::Session_label’ to ‘Genode::Number_of_bytes&’
.../repos/base/include/util/string.h:425:16: note: Genode::size_t Genode::ascii_to(const char*, double&)
  inline size_t ascii_to(const char *s, double &result)
                ^
.../repos/base/include/util/string.h:425:16: note:   no known conversion for argument 2 from ‘Genode::Session_label’ to ‘double&’
In file included from .../repos/base/src/test/printf/main.cc:18:0:
.../repos/base/include/util/xml_node.h: In member function ‘bool Genode::Xml_attribute::value(T*) const [with T = Genode::Session_label]’:
.../repos/base/include/util/xml_node.h:164:3: warning: control reaches end of non-void function [-Wreturn-type]
   }
   ^
make[3]: *** [main.o] Error 1
```
It seems the compiler does not pick `Xml_node::value(String<N> *out)` but `Xml_node::value(T *out)` although Session_label is a `String<160>`.

nfeske, 2016-08-19:
Even though the `Session_label` inherits `String<>` it apparently does not match for the C++ overload resolution. The solution would be to add a special overload of `ascii_to` to session_label.h, which matches the actual type.

ehmry, 2016-08-19:
I've found this works but it is hardly readable

``Session_label label = 		xml.attribute_value("label", String<label.capacity()>("nolabel")).string();``

nfeske, 2016-08-19:
Could you give the `ascii_to` overload idea a try? I am thinking along these lines:

```
inline size_t ascii_to(const char *s, Session_label &result)
{
  String<Session_label::capacity()> label;
  bool const ret = ascii_to(s, label);
  result = Session_label(label.string());
  return ret;
}
```

chelmuth, 2016-08-19:
There's no `ascii_to` overload/template for `String<>` (on master).

nfeske, 2016-08-19:
Ah, I see. This rings a bell: There cannot be a generic `ascii_to` overload for strings because this function would not know where the string ends. E.g. when requesting an attribute from an XML node, the string would and at the closing `"`. But in other situations, another token may separate the string from the rest. Hence, the extraction must happen somewhere where the information about the syntax around the string is known - such as the `Xml_node`. We seem have no other choice than to wrap @ehmry's variant into a convenience function, similar to the ones like https://github.com/genodelabs/genode/blob/master/repos/os/include/decorator/xml_utils.h

nfeske, 2017-02-28:
I'm closing the issue. To solve it, we needed to make `Xml_node` aware of `Session_label`, which is not good. In the last couple of weeks, I repeatedly encountered this problem and found the approach of using an intermediate `String` ok to work with:

```
typedef String<Session_label::capacity()> Label;
Session_label const label { xml_node.attribute_value("label", Label()) };
```

This works because the `Session_label` constructor accepts a `String<>` as argument.

closed 2017-02-28 by nfeske

chelmuth, 2017-02-28:
Agreed, the workaround looks fairly lean.