genodelabs/genode #2066: Running demo fails with : [init -> fb_sdl] Segmentation fault (signum=11), see Linux kernel log for details

ValentinQUEQUET, 2016-08-20:
genode build completed
spawn ./core
Genode 16.05
int main(): --- create local services ---
int main(): --- start init ---
int main(): transferred 17592186044415 MB to init
int main(): --- init created, waiting for exit condition ---
[init] Could not open ROM session for module "ld.lib.so"
[init -> fb_sdl] Segmentation fault (signum=11), see Linux kernel log for details
blocking canceled in entrypoint constructor
[init -> report_rom] parsing legacy <rom> policies
[init -> report_rom] parsing legacy <rom> policies
[init -> nitpicker_config] top-level node <xray> missing in input ROM xray
[init -> nitpicker_config] could not obtain input value for input xray_enabled
Expect: 'interact' received 'strg+c' and was cancelled

cnuke, 2016-08-21:
This problem should already be fixed on the current master branch — please give it a try.

ValentinQUEQUET, 2016-08-22:
Hello Team,

I very don't know why at all, but, when trying the master branch I just fetch over git, I got another error :

./tool/create_builddir build.linux
make: ***  Aucune règle pour fabriquer la cible « /home/valentin/OperatingSystems/Genode/genode-master/genode/tool/builddir/etc/build.conf.build.linux », nécessaire pour « /home/valentin/OperatingSystems/Genode/genode-master/genode/build/build.linux/etc/build.conf ». Arrêt.

And this whether I specify a build dir name or not.

This seems very odd to me ; what do you think about this ?

I am going to investigate that matter, the very first spare time I have to spent on it.

Thanks in advance for digging into it :-)

Sincerely,
Valentin

chelmuth, 2016-08-22:
You specified `build.linux` which is an invalid target platform. Just run the tool without arguments to get a list of valid target platforms. I expect you're looking for `linux_x86`.

ValentinQUEQUET, 2016-08-22:
Hi (again),

Don't bother scratching your head, I did the following mistake :

I voluntarily deleted the "./build/" dir I believed was the effective build dir instead of "./build/linux_x86" .

My appologies for the noise.

Courage, the framework needs testers ...

Sights,
Valentin

ValentinQUEQUET, 2016-08-22:
To Chelmuth :>\

You're completely right ; how can I have done such a stupid error ?

Besides this true proven fact, I checked my logs and I did this mistake only once while doing many tries which where rather proper (mistaken the right build dir).

So my first, initial, mistake still is at fault.

Very much thanks for your help ; good catch Sir.

To serve you Sir (and the whole Team),
Valentin

chelmuth, 2016-08-22:
For clarification: Does the issue still persist with the master branch? Please provide a log in this case.

ValentinQUEQUET, 2016-08-22:
To CNuke :>\

I feel uncomfortable ; who do you want to nuke ?
Maybe the C programming language :-)))

I can confess that you found the right solution to my problem, and as such, this bug can be closed right now cauz' the 'app' runs fine sine I switched to 'master' branch.

See you (or never),
Valentin

closed 2016-08-22 by chelmuth

ValentinQUEQUET, 2016-08-22:
To Chelmuth :>\

I very much hope you caught my last comment which reassures the community about the above-mentioned bug.

See you (next bug-time),
Valentin