genodelabs/genode #2072: os: abandon Irq_activation and Irq_handler deprecated

skalk, 2016-08-24:
Since the resolution of issue #1456, which turned the IRQ session interface into an asynchronous interface, there is no need for additional threads that wait for interrupts. Therefore, the helper classes `Irq_activation` and `Irq_handler` aren't necessary anymore. A simple `Signal_handler` should be enough. I would vote for marking the corresponding file `os/irq_activation.h` as being deprecated.

nfeske, 2016-08-24:
Very good suggestion!

skalk, 2016-08-24:
@nfeske above commit series solves the issue

nfeske, 2016-08-24:
Since the header produces a compile error if included, we may better remove all other content. The `#error` makes the code unusable anyway.

Furthermore, I noticed that `grep -ri  "Irq_activation" repos` still yields a few results. They should be adapted before we merge the patch:

* os/src/drivers/nic/spec/lan9118/lan9118.h
* os/src/drivers/nic/gem/cadence_gem.h
* os/src/drivers/sd_card/spec/pl180/pl180.h

skalk, 2016-08-24:
@nfeske I think you missed the other commits that reference the issue. They adapt the corresponding components.

skalk, 2016-08-25:
@nfeske: fixup 5a50d92377f78299b3f03351d170622880776d37 removes the content from the header as desired ;-)

commit f3edf82c (2016-08-25): os: remove Irq_activation and Irq_handler

Fix #2072
closed 2016-08-30 by chelmuth