genodelabs/genode #2082: Page-faults in thread context area invisible

skalk, 2016-08-31:
On the kernels: SeL4, Pistachio, Fiasco, OKl4 we do not see when a thread issues a page-fault due to a stack overrun. On Fiasco.OC there is no identity shown, but at least the page-fault address and instruction pointer. On HW and NOVA you can see the page-fault (including thread identity) in differently layouted messages.

It would be very nice to have corresponding messages on all kernels. It would be even nicer to unify the page-fault reporting.

chelmuth, 2016-09-08:
I really like this patch and it will enter staging in a second, but I still have one additional concern: Could you please have a look at *repos/base/src/core/region_map_component.cc* lines 171, 189, and 225? It seems their exists a platform-specific `print_page_fault` utility, which looks rather different on various kernels. It would be great to have a fixup commit that unifies these log outputs too.

skalk, 2016-09-08:
@chelmuth thanks for the review and improvement proposal. Fixup 23afcfc42f3444b366c8813f28ed88779d91cef8 tries to unify also the `print_page_fault` utility.

commit 0219d475 (2016-09-08): base: unify page-fault notification for kernels

* Enable page-fault messages for regions not directly managed by core
* Unify output given when a page-fault occurs related to those regions

Fix #2082
commit 2147c42e (2016-09-09): base: unify page-fault notification for kernels

* Enable page-fault messages for regions not directly managed by core
* Unify output given when a page-fault occurs related to those regions

Fix #2082
closed 2016-09-09 by chelmuth