genodelabs/genode #2084: LXQt Desktop

maverick74, 2016-08-31:
Since Genode is getting a LOT better and that it already supports Qt5 (THANK YOU!!!)

I think it would be great to have LXQt running on it - it's light, based on Qt, well known and would allow Genode to get to more users/testers )

chelmuth, 2016-08-31:
The first question that came to my mind while reading your posting was: Is LXQt really portable or is it tightly knit to Linux (or *BSD) like other desktop environments?

ssumpf, 2016-08-31:
@maverick74: Great suggestion, but we are still eight people, that try to build up an operating system/community. As @chelmuth implies, we are glad about any suggestions or evaluation efforts. Any help is welcome, so maybe I could invite you to investigate if this project is feasible on Genode. That would be awesome!

maverick74, 2016-09-05:
Hi
Ok, after asking for some help i was told by one of the lead developers - PCMan - that:

> LXQt requires some *nix features, such as X11 and dbus.
> Ideally Qt should be portable, but Qt-alone is far from enough to build a DE.
> At least there must be X11 and POSIX system calls on the target platform.
> That's the minimal requirements.

He sent a few more details on a second e-mail:

> 1. The main components using X11 are lxqt-session, lxqt-panel and the
> desktop icons / wallpaper parts of pcmanfm-qt. Other major components
> do not rely on X11.
> 2. Currently we're using openbox as the default window manager and
> kwin is a nice alternative. However, for a GUI without X11 and a WM,
> some efforts might be needed to get things work. All existing DEs have
> this problem, though.
> 3. lxqt-polkit uses polkit, which is Linux-only, but it's an optional component.
> 4. lxqt-admin uses polkit and systemd, but again, it's an optional component.
> 5. lxqt-session might optionally use libudev, which is linux-only, but
> this can be turned off with cmake flags.

As far as i know Wayland support is planned however, from the informations i have, it will depend on the use of KDE's Kwin...

Hope this helps!

It would be very nice to have LXQt on Genode!!! (in fact i believe it would be good for everyone if Genode could reuse other already existing projects such as LXQt, Flatpak, Wayland, CUPS, ALSA, etc. It would save man power and allow to focus on what is really important and that differentiate Genode. But, then again and despite my good will, truth be told that **i have no idea of what i'm talking about** ;P)

maverick74, 2016-12-02:
So, did it help?

is there something else i can do (beside coding in which i'm very limited...)?

nfeske, 2016-12-03:
Raising awareness about related projects is certainly a good intention but in my experience, it rarely has a lasting effect unless actual developers become *really* interested in the proposed direction. In most cases, active developers already have more plans to pursue than time to spend. Even though one finds a certain direction attractive, there is not enough incentive to overthrow existing priorities.

The two most effective ways to move things forward are becoming a developer, or creating new incentives for developers that are more profound than mere suggestions. The latter (like fostering a community, acquiring funding, forming a company, organizing events, advocating to potential future developers) is even harder than the former. ;-)

nfeske, 2017-12-13:
I am closing this issue as there are not plans to actively work on it.

closed 2017-12-13 by nfeske

ehmry, 2018-08-31:
Hi @maverick74,

The sculpt manager has a menu and something like task status views. One or two clock apps have been made, but because we usually have at least one VM running on our machines, adding a native clock feels a bit cluttered. When support for natively managing X windows from VMs is released and guest desktops disappear, I expect this to be an active topic.

nfeske, 2018-08-31:
@maverick74, the idea about a community forum popped up earlier this year on our mailing list. Please have a look at the discussion at https://lists.genode.org/pipermail/users/2018-January/005799.html

Right now, the primary point of communication about Genode is still the mailing list (https://genode.org/community/mailing-lists). Please do not hesitate to join.

maverick74, 2018-09-03:
Hi @ehmry 
From your description i get the feeling it's not based on the "traditional concept"...

About "support for natively managing X windows" is that going to be really **X** (as in X.org), wayland, or something else?

is that supposed to happen sometime in 2019?

Hi @nfeske 
well, it's good to know i wasn't the first to think about it ahahaha
thank you for the links and the invitation.
I really like to follow genode as close as i can. Shame i can't do any "hard coding".

@nfeske @ehmry  thank you both for your replies :)

ehmry, 2018-09-04:
@maverick74 native X/Wayland/Mir should never happen, but [Xpra](https://en.wikipedia.org/wiki/Xpra) is in the works.