genodelabs/genode #2086: core: page faults during rm_faulter::dissolve_from_faulting_region_map

alex-ab, 2016-09-05:
nova_x86_32 in qemu - run/platform

```
89 MiB RAM assigned to init
[init] Error: Could not open ROM session for "ld.lib.so"
[init -> test-platform] testing base-nova platform
[init -> test-platform] Warning: pager: created
[init -> test-platform] mapper: hello
[init -> test-platform] mapper: request mapping 0x9000
[init -> test-platform] Error: pager: got map request
page fault, pd='init -> test-platform' thread='mapper' cpu=0 ip=0x100091c address=0x9000

PAGE-FAULT IN CORE (READ pf_addr=6d0d4 pf_ip=15bc98 from 00 'unknown':'unknown')

stack pointer 0xa01febb0, qualifiers 0x4 irUwp
faulter utcb 0xa01ff000, last message item count 0
  #1 0xa01febb0 0x0015bc98
  #2 0xa01febbc 0x0016586d
  #3 0xa01febfc 0x0013c64e
  #4 0xa01fed3c 0x00104103
  #5 0xa01fee3c 0x001072c8
```

alex-ab, 2016-09-05:
Please have a look, the "core: ..." issue solves the issue for me.

alex-ab, 2016-09-05:
Hm, the fix avoids core to crash but scenarios like ```run/rm_nested``` don't work anymore.

alex-ab, 2016-09-06:
56cec43c474242b3ba256f80a0e25517747ea806 avoids the core crash and let the nested region test pass - as stop gap solution

chelmuth, 2016-09-09:
@alex-ab which commits of your topic branch are ready for staging?

alex-ab, 2016-09-09:
474e982a01009c6ffc82ab2c5450e49f5ff90bba and 39d09150e0ca9f708f4438312baf98d618d67ac6

alex-ab, 2016-09-09:
f47feeb696759435702312ea2a18de8a35f5079f contains the suggested weak_object and weak_ptr version.

nfeske, 2016-09-09:
Yes! :-)

chelmuth, 2016-09-12:
Merged as 4629371.

alex-ab, 2016-12-08:
@chelmuth, @nfeske: please consider 6bd0e48938f17872cc45f7f7805485a24ccbf227 for staging (and 47d377a5a0b606cfb84245c98617800969cbe1c4). The commit on master 99fbb23ec53bd5f10038890c4d9986613b4548f7 which uses Weak_ptr/Weak_object (as discussed here) in a wrong way (re-)introduced  a deadlock in core.

chelmuth, 2016-12-08:
Nice, merged both commits into staging.

commit 4629371d (2016-09-12): core: use weak_ptr for Rm_faulter and Region_map

Fixes #2086
commit 99fbb23e (2016-09-14): core: use weak_ptr for Rm_faulter and Region_map

Fixes #2086
closed 2016-12-14 by alex-ab