genodelabs/genode #2091: hw: let core/kernel be part of the address space of all components

skalk, 2016-09-08:
This is primarily to limit execution overhead when having syscalls, exceptions, or signaling/messaging. Currently, every kernel entry/exit implies a context switch. It implies a probably heavier TLB overhead in contrast to globally valid kernel pages. Moreover, the current design complicates certain architecture-specific operations, like limited cache flushing for specific protection domains, which is thereby costly too. To sum it up, the hardware designers are assume a design, where kernel and userland share somehow the address space view. By following an unfamiliar design we have to pay the costs.

The implication of running core/kernel within the same address space like user-land components is that core needs to be executed in privileged mode too. As core is sharing the address-space with its privileged kernel library anyway, I cannot see a security problem hereby. Core is part of the TCB of all components, like the kernels it runs on top of. Especially, it does not differ security-wise from the limited HW kernel, whose memory is managed by core.

skalk, 2017-10-04:
@chelmuth I've finally fixed the issue. All commits necessary are merged to current staging, and are available on my issue branch called issue#2091.

m-stein, 2017-10-12:
Despite some minor comments that I made, I totally enjoyed reading the commits. Especially that they reduce and simplify the hardly maintainable assembler code of the base-hw repository! Thanks!

chelmuth, 2017-10-13:
I added 2f51e8bce16ffa340caef77811e3eccc407bd67f to fix builds with `-O0`. Unfortunately, the generated binaries work only for hw/x86_64 but not on pbxa9. My bisect points to either d02072fcfa613580d25768699d0c2e8d57a01489 or 3f1b52a2df8da95b8bdbd0f9e94e2079ab514d8f as the first bad commit. The latter hangs after `kernel initialized` and all later commits hang after init was started. Please note that each bisect steps needs `git show 2f51e8bce16ffa340caef77811e3eccc407bd67f | patch -p1` to fix building `-O0`.

skalk, 2017-10-16:
@chelmuth: I've added an interims solution with fixup 8d99d5c . Therewith, -O0 should compile and run again. A more nice solution is already part of some cleanup work that is already in the pipe.

commit d02072fc (2017-10-12): hw: switch page-tables only when necessary

* Instead of always re-load page-tables when a thread context is switched
  only do this when another user PD's thread is the next target,
  core-threads are always executed within the last PD's page-table set
* remove the concept of the mode transition
* instead map the exception vector once in bootstrap code into kernel's
  memory segment
* when a new page directory is constructed for a user PD, copy over the
  top-level kernel segment entries on RISCV and X86, on ARM we use a designated
  page directory register for the kernel segment
* transfer t [... cut]
closed 2017-10-19 by chelmuth