genodelabs/genode #2092: hw: bootstrap the kernel into virtual memory

skalk, 2016-09-08:
To put issue #2091 into practice, it would be nice to split the functionality of bootstrapping the hardware into virtual memory mode (enable the MMU) and the rest of the kernel/core. Due to the fact that enabling the MMU needs a lot of things to be done, dependent on the concrete hardware, as well as creating core's page tables can be complex and is platform-dependent too, a lot of code needs to be shared between bootstrap code and the rest of the kernel.
On the other hand, the bootstrap code must be linked to physical memory addresses, and the rest of the kernel to virtual ones. When we do not want to duplicate code, and to not be restricted to only use macros, or inline functions as common code base, we should split the bootstrap functionality into an extra binary. Thereby, it can use the same C++ classes etc. like the kernel. It loads core, prepares core's page tables, enables the hardware appropriately, and then starts core in privileged mode with the MMU enabled.

skalk, 2017-02-02:
@nfeske @chelmuth above commits 61291dc to dc15da7 implement the bootstrapping issue. You can also refer to my topic branch called issue#2092 that is freshly rebased to current staging.

nfeske, 2017-02-07:
@skalk A feast for the eyes! Thank you for this beautiful commit series. I merged is to staging now.

nfeske, 2017-02-28:
This is fixed in Genode 17.02. @skalk How about closing the issue?

skalk, 2017-03-01:
sure!

closed 2017-03-01 by skalk