genodelabs/genode #2094: Make GNU build-system wrapper usable for non-Noux programs

nfeske, 2016-09-11:
The ports/mk/noux.mk wrapper for the GNU build system is very handy to port software that is based on the GNU build system to the Noux runtime. However, there are many applications that come readily equipped with support for the GNU build system but don't require actual Unix features (i.e., fork). Right now, porting such applications requires us to express their build rules via the Genode build system. In most cases, this is fairly easy but still requires manual labor. It would be nice to be able to use the build-system wrapper as used by noux packages for such applications as well.

chelmuth, 2016-09-19:
The build fails with
```
  CONFIG   vim
Error: configure script of vim failed
.../contrib/vim-460a4a30deb93fee494a17ff6e64afaf813a37dd/src/noux-pkg/vim/configure: 6: cd: can't cd 
```
Is this related to d603adc?

nfeske, 2016-09-20:
Yes, thanks for the notice. I added a fixup for the original commit.

nfeske, 2016-10-05:
Commit c500f6657c415c0ef8cd05f84576c29079a20519 should fix the issue along with the warnings during the clean rule about the disappearing current working directory.

commit d603adc2 (2016-09-15): ports: GNU build-system support for non-noux pkgs

This patch makes the build-system integration of noux packages usable
for non-Noux targets. It moves the GNU build system wrapper to
ports/mk/gnu_build.mk, which is now included by noux.mk. This way,
non-noux applications can use the same build-system wrapper. So the
porting of individual applications becomes easier.

This change removes the 'NOUX_' prefix use the various build variables
used by the noux-pkg's target.mk files to steer the behaviour of the GNU
build system. E.g., NOUX_CONFIGURE_ARGS is now called CONFIGURE_ARGS.
Note that there [... cut]
commit e806c8ea (2016-09-30): ports: GNU build-system support for non-noux pkgs

This patch makes the build-system integration of noux packages usable
for non-Noux targets. It moves the GNU build system wrapper to
ports/mk/gnu_build.mk, which is now included by noux.mk. This way,
non-noux applications can use the same build-system wrapper. So the
porting of individual applications becomes easier.

This change removes the 'NOUX_' prefix use the various build variables
used by the noux-pkg's target.mk files to steer the behaviour of the GNU
build system. E.g., NOUX_CONFIGURE_ARGS is now called CONFIGURE_ARGS.
Note that there [... cut]
commit 7bed3967 (2016-10-21): ports: GNU build-system support for non-noux pkgs

This patch makes the build-system integration of noux packages usable
for non-Noux targets. It moves the GNU build system wrapper to
ports/mk/gnu_build.mk, which is now included by noux.mk. This way,
non-noux applications can use the same build-system wrapper. So the
porting of individual applications becomes easier.

This change removes the 'NOUX_' prefix use the various build variables
used by the noux-pkg's target.mk files to steer the behaviour of the GNU
build system. E.g., NOUX_CONFIGURE_ARGS is now called CONFIGURE_ARGS.
Note that there [... cut]
closed 2016-10-21 by chelmuth