genodelabs/genode #2095: core: unify handling of boot modules

skalk, 2016-09-13:
Currently, we assemble all elf-binaries that should be available after booting within memory (core's ROM modules) into one elf-binary when running on top of the HW kernel or SeL4. Moreover, the process of assembling is duplicated across those both platforms. It would be nice to unify this approach.

From my point of view, it would be best to follow this approach across all supported kernel platforms, to reduce platform-specific code even more.

skalk, 2016-09-13:
To achieve above described goal, it is necessary to build core as library on all platforms, and then link the final core image within the run-tool, when all boot modules are available. All special things done within the core target descriptions then have to move into the run-tool, e.g., special linker addresses on Fiasco.OC, or special linker-file on NOVA.

skalk, 2016-09-23:
I''l remove the fixed label, because it breaks vbox_auto* autopilot tests for hw_x86_64_muen

skalk, 2016-09-28:
@Kensan @reet- : would you be so kind to give my branch: https://github.com/skalk/genode/tree/issue%232095 another try with respect to hw_x86_64_muen? I have some assumption regarding the broken autopilot tests.

reet-, 2016-09-28:
@skalk Vbox tests fail, see https://gist.github.com/reet-/c34085f37a5b60bfdf64a70bdc4fdf81

skalk, 2016-09-29:
@reet- ok, thanks for testing. Nevertheless, that is pretty strange. I have no clue at all, how my changes should relate to this VBox error. In fact the only thing that changed is the binary layout of core. No single bit changed in anything else, e.g., base-library. Thar is why I'm out of ideas how Vbox/Muen breaks.

reet-, 2016-09-29:
@skalk You changed almost nothing, only the binary layout of core ;) 

`vbox_auto_win7` shows the exact same behavior on `nova_x86_64`.

skalk, 2016-09-29:
@reet- ah ok, thanks for the pointer!

reet-, 2016-09-29:
@skalk Np, please let me know if I should test anything.

skalk, 2016-09-29:
@reet- I cannot reproduce the problem with nova_x86_64. The `vbox_auto_win7` test runs fine with my branch.

skalk, 2016-09-29:
@reet- ok, sorry forget about my last comment. Now i have pushed a fixup for the try-out commit to the same branch. Would you be so kind to give it another try?

reet-, 2016-09-29:
@skalk Ok we are back to the initial failure (`VERR_GENERAL_FAILURE TCG`, see https://gist.github.com/reet-/7b2cfb134b011d8a8180f60d9468a311).

skalk, 2016-09-30:
@reet- : Hi, I've a suspicion what might be the difference to the working version. Can you please put the following line into `repos/base-hw/mk/spec/hw_x86_64.mk`:

```LD_TEXT_ADDR ?= 0x200000```

reet-, 2016-09-30:
@skalk You just saved my day. Works!

Strange, I checked the layout of the text segment yesterday because this line looked suspicious, but I could not see any obvious difference. Furthermore all scenarios besides vbox worked fine.

Can you elaborate?

skalk, 2016-09-30:
I would say the change in the virtual memory layout of virtualbox only revealed a memory-issue in the virtualbox/muen port. My original commit unified the link address for all platforms to 0x01000000.

reet-, 2016-09-30:
@skalk Ok thanks for the pointer, investigating.

skalk, 2016-09-30:
@chelmuth ok, the final version is online

reet-, 2016-10-03:
We investigated the problem on Muen and it turns out that the QEMU TCG code makes assumptions about the relation of virtual memory which are not met any longer.

See the `tcg_out_modrm_sib_offset` function in `src/app/virtualbox/src/recompiler/tcg/i386/tcg-target.c`. The `disp` variable is calculated from a given `offset` and `code_ptr`. The actual values of these variables depend on allocation order and available virtual memory.

On hw_x86_64_muen, `code_ptr` has a low value (0x800000) while `offset` holds a big number (0xf9000060). The calculation

`tcg_target_long disp = offset - pc;`

then yields a result which the TCG code deems as not addressable:

```
[init -> vbox] EMT      emulate_single_instr
[init -> vbox] EMT      tb_gen_code phys_pc 0xc11e4 pc 0xc11e4 cs_base 0xc0000 tc_ptr 0x800000
[init -> vbox] EMT      tcg_gen_code
[init -> vbox] EMT      tcg_gen_code_common code_ptr 800000
[init -> vbox] EMT      tcg_out_modrm_offset opc 2187 r 5, rm 14, offset 8
[init -> vbox] EMT      tcg_out_modrm_sib_offset opc 2187 r 5, rm 14, index -1, shift 0, offset 8
[init -> vbox] EMT      tcg_out32 code_ptr a975fb
[init -> vbox] EMT      INDIRECT jump to target f9000060
[init -> vbox] EMT      tcg_out_modrm_offset opc 255 r 4, rm -1, offset f9000060
[init -> vbox] EMT      tcg_out_modrm_sib_offset opc 255 r 4, rm -1, index -1, shift 0, offset f9000060
[init -> vbox] EMT      tcg_out_modrm_sib_offset code_ptr 800015 tcg_calc_opc_len 1
[init -> vbox] EMT      tcg_out_modrm_sib_offset pc 80001b
[init -> vbox] EMT      tcg_out_modrm_sib_offset disp f8800045
[init -> vbox] EMT      tcg_out_modrm_sib_offset disp == (int32_t)disp 0
[init -> vbox] EMT      internal REM fatal error: rc=VERR_GENERAL_FAILURE TCG
```

If I fiddle with the allocation order (see [here](https://git.codelabs.ch/?p=genode.git;a=commitdiff;h=14f7b4f836e0ad5e4e7754e00cc1d8e085f32c61)), the vbox scenarios on Muen work again.

Before the boot module changes, the memory layout and allocation order resulted in a `disp` value which, more or less by chance, fits into `int32_t`.

The constraint on the code generation buffer is mentioned in `src/app/virtualbox/src/recompiler/exec.c`, line 556:

`/* The code gen buffer location may have constraints depending on the host cpu and OS */`

What would be the best way to address this issue? As we see it, this could happen on other platforms as well.

alex-ab, 2016-10-03:
Good catch ! The next time this base + offset thing of vbox bite us ;-( . In repos/ports/src/virtualbox/mm.cc we already use managed dataspaces to make sure that allocation of a specific type are within a 1 << 31bit window. Maybe we should/can drop the rt.cc specific implementation and switch to the one/reuse the mm.cc version. Would this solve the issue from your perspective @reet ?

reet-, 2016-10-04:
The description of the `Sub_rm_connection` class sounds like it would indeed solve the issue we observed. @alex-ab Could you provide a try-out commit to make sure?

alex-ab, 2016-10-04:
@reet-: may you give commit 231cdb21640a8abf81d73add577d168c5a95287c a try ?

reet-, 2016-10-04:
Works for us! Thanks @alex-ab.

skalk, 2016-10-10:
@nfeske @chelmuth I've rebased it once again to current staging

skalk, 2016-10-19:
@chelmuth commit 9f4d2a851c8700efab59203f44e73b28b0e5d027 is based on current staging

commit 23808c68 (2016-10-21): core: unify handling of boot modules

Instead of solving the problem to deliver ROM modules to core while booting
differently for the several kernels (multi-boot, elfweaver, core re-linking),
this commit unifies the approaches. It always builds core as a library, and
after all binaries are built from a run-script, the run-tool will link an
ELF image out of the core-library and all boot modules. Thereby, core can
access its ROM modules directly.

This approach now works for all kernels except Linux.

With this solution, there is no [build_dir]/bin/core binary available anymore.
For debugging pu [... cut]
commit 7e1692d9 (2016-11-08): core: unify handling of boot modules

Instead of solving the problem to deliver ROM modules to core while booting
differently for the several kernels (multi-boot, elfweaver, core re-linking),
this commit unifies the approaches. It always builds core as a library, and
after all binaries are built from a run-script, the run-tool will link an
ELF image out of the core-library and all boot modules. Thereby, core can
access its ROM modules directly.

This approach now works for all kernels except Linux.

With this solution, there is no [build_dir]/bin/core binary available anymore.
For debugging pu [... cut]
closed 2016-11-08 by chelmuth