genodelabs/genode #2103: fs_log test fails on hw_x86_64

chelmuth, 2016-09-15:
The issue can be reproduced in Qemu. It also seems to affect hw_x86_64_muen. Log follows
```
[2016-09-14 04:32:52] kernel initialized
[2016-09-14 04:32:52] Genode 16.08
[2016-09-14 04:32:52] 121 MiB RAM assigned to init
[2016-09-14 04:32:52] Warning: Quota exceeded! amount=118784, size=12288, consumed=110592
[2016-09-14 04:32:52] Warning: Insufficient quota for transfer: 
[2016-09-14 04:32:52] Warning:   have 8192, need 12288
[2016-09-14 04:32:52] Warning: not enough quota for a donation of 12288 bytes
[2016-09-14 04:32:52] [init] resource_request: ram_quota=12288
[2016-09-14 04:32:52] Warning: Insufficient quota for transfer: 
[2016-09-14 04:32:52] Warning:   have 0, need 12288
[2016-09-14 04:32:52] Warning: not enough quota for a donation of 12288 bytes
[2016-09-14 04:32:52] [init] resource_request: ram_quota=12288
[2016-09-14 04:32:52] [init] Error: Uncaught exception of type 'Genode::Parent::Quota_exceeded'
[2016-09-14 04:32:52] [init] Warning: abort called - thread: vfs
[2016-09-14 04:32:52] child "init" exited with exit value 1
[2016-09-14 04:32:52] [init -> timer] maximum timeout 549439000 us
[2016-09-14 04:36:52] Error: Test execution timed out
```

Kensan, 2016-09-19:
Side-node: decreasing the **upgrade_pd_session_quota** in **Genode::ipc_reply_wait** (see 4935e36...) lets the test pass on **hw_x86_64** as well as on Muen.

skalk, 2016-09-22:
@Kensan I think this is no fix in general. In this concrete example it might help, because init has coincidentally 8K ram quota left, but not 12K.

In general the problem here is that init needs dynamically more RAM (due to permanent capability reception), but it gave away everything to one of its children (bomb). Of course we can lower the quota donation, or lower the RAM given to the children, and then press our thumbs. The real problem is that activity of some clients (e.g., children) can lead to memory pressure of the PD session of the service.
If we do not want, or are not able to target this structural problem at the moment, I would limit the memory given to the bomb child right now. @chelmuth what do you think about this?

chelmuth, 2016-09-22:
When first discussing the issue with @Kensan some weeks ago I was thinking in the same direction. Would it help to just add `demand="2M"` to the bomb config node? @alex-ab?

skalk, 2016-09-23:
Ok, I think we found a short-term fix by increasing the slack RAM quota of the init component.

In general there are still two structural issues to be solved. The one is that init has to do certain dynamic allocations on behalf of its children, e.g., session capabilities that init preserves. On the other hand we have the generic problem that clients needs to donate RAM for capabilities to a service, which receives them while performing on behalf of the client.

The specific init issue might get relaxed, when init gets re-designed due to a more asynchronous parent interface @nfeske?

closed 2016-10-21 by chelmuth

ehmry, 2016-12-14:
I've had to bump this limit up again using a heeselicht based scenario, this time triggered by the decorator.

commit 77126d5b (2016-09-23): os: increase init's slack RAM quota (fix #2103)
commit 5d7e2f1a (2016-09-30): os: increase init's slack RAM quota (fix #2103)
commit 0cbfef7b (2016-10-21): os: increase init's slack RAM quota (fix #2103)