genodelabs/genode #2104: hw: assertion in cpu_sampler test on hw_pbxa9

chelmuth, 2016-09-15:
```
[2016-09-14 04:15:34] kernel initialized
[2016-09-14 04:15:34] Genode 16.08
[2016-09-14 04:15:34] 761 MiB RAM assigned to init
[2016-09-14 04:15:34] [init] Error: Could not open ROM session for "ld.lib.so"
[2016-09-14 04:15:34] [init -> timer] maximum timeout 4294967000 us
[2016-09-14 04:15:34] [init -> cpu_sampler] starting a new sample period
[2016-09-14 04:15:34] [init -> init] Error: Could not open ROM session for "ld.lib.so"
[2016-09-14 04:15:34] Warning: Transfer CPU quota, init -> init, targeted session not found
[2016-09-14 04:15:35] Error: Assertion failed: _state == AWAITS_RESUME || _state == ACTIVE
[2016-09-14 04:15:35] Error:   File: /home/checker/checker.git/genode-package/repos/base/../base-hw/src/core/kernel/thread.cc:137
[2016-09-14 04:15:35] Error:   Function: void Kernel::Thread::_pause()
[2016-09-14 04:15:44] Error: Test execution timed out
```

m-stein, 2016-09-15:
I'll have a look at it. Thanks for the hint!

m-stein, 2016-09-15:
@chelmuth I've uploaded a fix.

chelmuth, 2016-09-16:
I removed the fixed label until the offline discussion settles.

chelmuth, 2016-12-01:
Now it seems my duty to ask: What was the outcome of the offline discussion?

m-stein, 2016-12-02:
@chelmuth I rebased it.

chelmuth, 2016-12-07:
The commit breaks the build for all platforms but arm.

m-stein, 2016-12-07:
@chelmuth Thanks for the hint! I have added a fixup.

closed 2016-12-14 by nfeske