genodelabs/genode #2105: Consistently define `Genode::size_t` as `unsigned long`

nfeske, 2016-09-15:
Currently, `Genode::size_t` is defined as `__SIZE_TYPE__` (in https://github.com/genodelabs/genode/blob/master/repos/base/include/base/stdint.h), which is provided by the compiler. The type should match the machine-register width because it is typically used for address calculations. On x86_32, this is the case for `unsigned int` so `__SIZE_TYPE__` is an `unsigned int` when building for this architecture. However, on x86_64, `unsigned int` is a 32-bit integer. So it is not suitable for a size type. Therefore, when compiling for this architecture `__SIZE_TYPE__` is an `unsigned long`.

This becomes a problem with my current ambition to make library symbol maps a formal part of the library API (in addition to the headers of the library). All symbols for C++ methods or functions that take a `Genode::size_t` as argument look different on x86_32 and x86_64, which defeats the idea to have architecture-agnostic symbol maps.

By consistently defining `Genode::size_t` as `unsigned long`, we could solve this problem. On both 32 and 64 bit, `unsigned long` has the same width as machine words.

chelmuth, 2016-09-15:
I totally support this change but it comes with a good deal of work in places where `Genode::size_t` is used like ANSI `size_t`, i.e., in format string functions, which will produce a warning diagnostic. With our current ambitions all these places will be outside of the base framework, e. g., in ported components and libraries. So, let's go for it.

nfeske, 2016-09-15:
I see the concern and hope that 3rd-party code that relies on format strings will use the libc for that (this is plausible as Genode won't have format-string functions in the future). So chances are good that such code will use the `size_t` provided by the libc, which remains to be consistent with the compiler's `__SIZE_TYPE__`.

nfeske, 2016-09-15:
I merged the commit above to staging. Hopefully I have found most places where the change causes build errors. We should keep an eye on the compile warnings now. ;-)

commit 64093288 (2016-09-15): Define Genode::size_t as unsigned long

Fixes #2105
commit d96eda36 (2016-09-30): Define Genode::size_t as unsigned long

Fixes #2105
commit e370e08e (2016-10-21): Define Genode::size_t as unsigned long

Fixes #2105
closed 2016-10-21 by chelmuth