genodelabs/genode #2106: Remove placement new operators

nfeske, 2016-09-15:
We should use the `construct_at` function (util/construct_at.h) wherever possible.

The biggest offenders are the net headers at `os/include/net`.

chelmuth, 2016-09-19:
Don't forget `repos/os/src/drivers/platform/spec/rpi/*.h`.

trimpim, 2016-09-20:
I will rebase my IOMUX patch to the newest framework next week, I could do this then.

nfeske, 2021-03-05:
Commit https://github.com/genodelabs/genode/commit/f26cacad803bf4dd622b04a4a3a9540aabe254a8 adjusts the Rpi platform driver.

nfeske, 2021-03-05:
Commits https://github.com/genodelabs/genode/commit/57ebd947eda73e81080b594885cd3616454b1fb5 and https://github.com/genodelabs/genode/commit/36e7c64b448037fffc2bca357a907e82bc902c94 wrap up this issue.

commit 36e7c64b (2021-03-05): base-foc/hw: avoid use of placement new operator

Fixes #2106
closed 2021-03-12 by nfeske