genodelabs/genode #2108: dde_ipxe: increase RXD and TXD for Intel NICs

cnuke, 2016-09-17:
Using the `nic_drv` while AMT is enabled can lead to the PHY getting stuck for all but the AMT traffic. For that to happen it does not matter if AMT is actually used (SOL); it is enough to have it enabled. This behaviour was observed on a Thinkpad X250 which uses a I218-LM device. On a Shuttle DS57U which does not support AMT but also features the same device the issue does not exist.

iPXEs Intel driver sets the RX and TX descriptor number to low values (8 and 16). Increasing those to 16 and 32 solves the AMT issue. Although the spec gives indeed 8 as the minimum descriptor number, most other OS drivers, i.e., Linux and OpenBSD, use 256 as default.

I have adjusted the values accordingly and tested it on various systems (X220, X250, DS57U, macho) and they work fine.

ssumpf, 2016-09-18:
@cnuke: :+1:

chelmuth, 2016-09-18:
Thanks for your extensive investigation!

commit 0042a34f (2016-09-18): dde_ipxe: increase RXD and TXD for Intel NICs

The default iPXE settings of 8 (RXD) and 16 (TXD) lead to issues when
AMT is enabled. In most other OS drivers those descriptors are set
to 256, let us do the same. This should fix all observed issues wrt.
AMT.

Fixes #2108.
commit 35901c9c (2016-09-30): dde_ipxe: increase RXD and TXD for Intel NICs

The default iPXE settings of 8 (RXD) and 16 (TXD) lead to issues when
AMT is enabled. In most other OS drivers those descriptors are set
to 256, let us do the same. This should fix all observed issues wrt.
AMT.

Fixes #2108.
commit 27491fe7 (2016-10-21): dde_ipxe: increase RXD and TXD for Intel NICs

The default iPXE settings of 8 (RXD) and 16 (TXD) lead to issues when
AMT is enabled. In most other OS drivers those descriptors are set
to 256, let us do the same. This should fix all observed issues wrt.
AMT.

Fixes #2108.
closed 2016-10-21 by chelmuth