genodelabs/genode #2109: hw: remove x86 vs. x86_64 abstraction

skalk, 2016-09-22:
As we only support 64 bit version of x86, and I do not see that this will change in the near future, it is annoying to maintain the additional abstractions when differentiating between common x86 and x86_64 specific code within `base-hw` (with Muen there is another dimension in this matrix). I'm not sure whether the differences between common x86 and x86_64 are correctly maintained until now.

Before starting to maintain the next x86* makefiles for a base-hw specific bootstrap component (issue #2092) I'd like to get rid of the x86 specifier in `base-hw`, and use x86_64 all along.

skalk, 2016-09-22:
@Kensan @reet- do you agree with those changes?

reet-, 2016-09-22:
@skalk While the change looks fine, vbox_auto* autopilot tests on hw_x86_64_muen fail (see log at https://gist.github.com/reet-/bd808c7fe8076c49743c5c7ab52ae7b1)

chelmuth, 2016-09-22:
Please add the fixed label again if the remaining issues are sorted out.

skalk, 2016-09-23:
@reet- It is hard for me to reproduce the vbox* scenarios on hw_x86_64_muen. Could you please check, whether the failure is related to above commit, or to 1a80392e1b3d6d532a09456717398d3cfcca4b71 or dfc5a14c50452e8e60797d6bc7ba021000d629e9
That would help me a lot.

reet-, 2016-09-23:
@skalk Yes sure, we will look into it this afternoon.

Kensan, 2016-09-23:
@skalk Commit 1a80392 still works while commit dfc5a14 contains the error.

skalk, 2016-09-23:
@kensan thanks for testing, I've removed the fixed label of the corresponding issue, and mark this work as being fixed again.

commit 1dffea4e (2016-09-23): hw: only use x86_64 specifier instead of x86

Fix #2109
commit c674beac (2016-09-30): hw: only use x86_64 specifier instead of x86

Fix #2109
commit 78410583 (2016-10-21): hw: only use x86_64 specifier instead of x86

Fix #2109
closed 2016-10-21 by chelmuth