genodelabs/genode #2115: base-core: rm_session::destroy hangs up

DenisHuberTUM, 2016-10-03:
When starting a simple component [1] which only creates and destroys a new Region map, the component hangs up while processing destroy.

[1] https://github.com/702nADOS/genode-CheckpointRestore-SharedMemory/blob/2bbedd561131123f433a66a385da38f53a23818e/src/test/rm_session_destroy/main.cc

I found out, that destroy issues Genode::destroy, which calls the dtor ~Region_map_component(), but never proceeds to the next command (in Genode::destroy). In fact, the dtor finishes all its commands, but the flow of the program does not return back to Genode::destroy.

alex-ab, 2016-10-04:
Following commit fixes the issue for me. @DenisHuberTUM: Thanks for reporting.

nfeske, 2016-10-06:
@alex-ab please see https://github.com/genodelabs/genode/commit/d54a56f5d7a8389301892c2535bc0ab6713284e1 ;-)

alex-ab, 2016-10-06:
I see ;-)

commit 07bc796a (2016-10-05): base: avoid deadlock in region_map destruction

Fixes #2115
closed 2016-10-21 by chelmuth