genodelabs/genode #2116: vbox: recompiler fails with "internal REM fatal error"

alex-ab, 2016-10-04:
```
[init -> vbox] EMT      internal REM fatal error: rc=VERR_GENERAL_FAILURE TCG fatal error: genode/contrib/virtualbox-9739dad12[0m
[init -> vbox] e66d0ac8a900e3c915ea34a1fea1c2b/src/app/virtualbox/src/recompiler/tcg/i386/tcg-target.c:514[0m
[init -> vbox] EMT      [0m
[init -> vbox] EMT      !!Assertion Failed!![0m
[init -> vbox] EMT      Expression: <NULL>[0m
[init -> vbox] EMT      Location  : genode/contrib/virtual[0m
[init -> vbox] box-9739dad12e66d0ac8a900e3c915ea34a1fea1c2b/src/app/virtualbox/src/recompiler/VBoxRecompiler.c(4802) remAbort[0m
[init -> vbox] [0m
[init -> vbox] !!Assertion Failed!![0m
[init -> vbox] Expression: <none>[0m
[init -> vbox] Location  : genode/contrib/virtualbox-9739dad12e66d0ac8a900e3c915ea34a1fea1c2b/src/app/virtualbox/src/recompiler/VBoxRecompiler.c(4802) remAbort[0m
[init -> vbox] EMT      internal REM fatal error: rc=VERR_GENERAL_FAILURE TCG fatal error: genode/contrib/virtualbox-9739dad12[0m
[init -> vbox] e66d0ac8a900e3c915ea34a1fea1c2b/src/app/virtualbox/src/recompiler/tcg/i386/tcg-target.c:514[0m
```
Discovered after changes due to #2095

alex-ab, 2016-10-04:
@reet-: please try the following commit again, it should basicly should do the same as the first version you tested in #2095. Thanks for bug hunting and testing again !

reet-, 2016-10-04:
@alex-ab Works.

alex-ab, 2017-03-11:
The issue occurs again. Unfortunately the REM (Qemu TCG) uses currently two different ways to allocate memory, once via RTMemExecAlloc (which we implement in rt.cc) and RTMemAlloc* which is backed finally by libc malloc and friends. The allocations on 64bit can be arbitrary large apart, so that this assertion in the REM triggers.

alex-ab, 2017-03-13:
Commit ac8bf858fdf564972a938ccea54ed6fa2db3f0e0 fixes the issue.

nfeske, 2017-12-13:
What is the status of this issue? It is marked as fixed but the patch by @alex-ab was not merged to master since a long time now.

alex-ab, 2017-12-14:
The patch caused some trouble on our nightly runs, but I had no time to investigate at that time.

alex-ab, 2019-03-14:
I re-based the commit and tested a bit. It is ready for a re-try on staging.

chelmuth, 2019-03-20:
Removed the commit from staging for further development.

alex-ab, 2019-03-21:
Reworked the commit, please schedule for testing next time.

nfeske, 2019-03-25:
@alex-ab I merged your commit to staging but I haven't managed to get a grip about that problem it addresses. Can you leave a short note about the circumstances of the problem?

nfeske, 2019-03-26:
The build of the vbox5-nova depot archive fails now:
```
/data/depot/genodelabs/src/vbox5-nova/2019-03-25/src/virtualbox5/rt.cc: In function âvoid* RTMemTCGRealloc(void*, size_t)â:
/data/depot/genodelabs/src/vbox5-nova/2019-03-25/src/virtualbox5/rt.cc:375:20: error: âRTMemTCGFreeâ was not declared in this scope
    RTMemTCGFree(ptr);
                    ^
/data/depot/genodelabs/src/vbox5-nova/2019-03-25/src/virtualbox5/rt.cc:400:18: error: âRTMemTCGFreeâ was not declared in this scope
  RTMemTCGFree(ptr);
                  ^
/data/genode/repos/base/mk/generic.mk:56: recipe for target 'rt.o' failed
make[6]: *** [rt.o] Error 1
```

@alex-ab Could you please have a look?

alex-ab, 2019-03-26:
@nfeske: please add ee04ccec3b7b7f573a547cd2fa57db8f3aaf846b as fixup

alex-ab, 2019-03-26:
> @alex-ab I merged your commit to staging but I haven't managed to get a grip about that problem it addresses. Can you leave a short note about the circumstances of the problem?

The original commit created per TCG allocation a dataspace, which leads to out of capability exceptions. As written in the commit, e.g. a 9GB VM required 2500 more caps then before the commit, so this was not going to work.

nfeske, 2019-03-26:
This part I understood. But I was missing the problem addressed by the original commit (when does the recompiler error occur?).

alex-ab, 2019-03-26:
The reason is, as mentioned in earlier commits, vbox stores pointers at some places in form of base + offset, whereby offset is int32_t. The assertion triggers, or more badly silent corruption occurs, whenever two allocations are more than the int32_t away (e.g. 1 << 31). Before the commit the libc mem alloc implementation was used for the REM compiler, where allocations can be arbitrary apart, e.g. on 64bit.

nfeske, 2019-03-26:
Thanks for the recapture, @alex-ab. Did you happen to have a way to stable way to reproduce the problem (like a specific vbox config)? I'm asking because I have (luckily) never experienced it myself.

alex-ab, 2019-03-26:
I'm personally had not this issue, but please read the referenced #2095 issue which lead to this issue. The anecdotes showcases quite nice the trouble you may get.

nfeske, 2019-03-26:
Oh yes. Thanks for the pointer, and for the remedy!

commit f8960ead (2016-10-05): vbox: alloc memory of RTMem*Alloc within 2G window

Fixes #2116
commit 07dd7e37 (2017-03-15): vbox: allocate REM memory dense

Fixes #2116
commit 976230e9 (2019-03-19): vbox: allocate REM memory dense

Fixes #2116
commit 30099861 (2019-03-25): vbox: allocate REM memory dense

Fixes #2116
closed 2019-04-02 by nfeske