genodelabs/genode #2117: vbox: handle initial link state in network backend

cnuke, 2016-10-06:


trimpim, 2016-10-07:
For me this patches crash vbox when booting in to a 64 bit Ubuntu 14.04 or Windows 7. A 32 bit Windows boots fine.
The error is:
page fault, pd='[init -> vbox]' thread='vCPU dispatcher' cpu=1 ip=0x13ce620 address=0x8

chelmuth, 2016-10-18:
I removed the `fixed` label because of @trimpim's post.

alex-ab, 2016-10-26:
Some more information. It may happen that, with this patch, the link state change is processed before the network model and the used data structure is ready initialized. The phenomena is that TMTimerSetMicro is called from DevE1000.cpp with a pTimer of null - which is normally initialized in e1kR3Construct. Somehow it must be take sure, that the network model is ready before sending link state to it (or delaying the registration of the signal handlers up to the point where it is known that the network model is ready). Please have a look.

alex-ab, 2016-10-27:
Please try following commit.

chelmuth, 2016-11-02:
Does c33d4fb fix the issue?

alex-ab, 2016-11-02:
Yes, as far as i know.

trimpim, 2016-11-03:
Yes, the issue is fixed with this patch.

commit 20726ffd (2016-11-08): vbox: set initial link-state

VirtualBox mainly derives the initial link-state for its device models
from checking the <Adapter ... cable="true"/> attribute. Our backend
only propagates the current state of the Nic session if it receives a
link-state signal. This may lead to problems if a guest detects a link
up state when it is actually down and wants to use the interface. The
backend now queries the Nic session and sets the link-state accordingly
when it is constructed.

Fixes #2117.
commit 4ef4076b (2016-11-09): vbox: improve network backend

VirtualBox mainly derives the initial link-state for its device models
from checking the <Adapter ... cable="true"/> attribute. Our backend
only propagates the current state of the Nic session if it receives a
link-state signal. This may lead to problems if a guest detects a link
up state when it is actually down and wants to use the interface. The
backend now queries the Nic session and sets the link-state accordingly
when it is constructed.

In case there is no link do not attempt to submit a packet to the packet
stream but return with an error so that upper la [... cut]
commit 741abb42 (2016-11-25): vbox: improve network backend

VirtualBox mainly derives the initial link-state for its device models
from checking the <Adapter ... cable="true"/> attribute. Our backend
only propagates the current state of the Nic session if it receives a
link-state signal. This may lead to problems if a guest detects a link
up state when it is actually down and wants to use the interface. The
backend now queries the Nic session and sets the link-state accordingly
when it is constructed.

In case there is no link do not attempt to submit a packet to the packet
stream but return with an error so that upper la [... cut]
commit 9ba7b2ed (2016-11-25): vbox: improve network backend

VirtualBox mainly derives the initial link-state for its device models
from checking the <Adapter ... cable="true"/> attribute. Our backend
only propagates the current state of the Nic session if it receives a
link-state signal. This may lead to problems if a guest detects a link
up state when it is actually down and wants to use the interface. The
backend now queries the Nic session and sets the link-state accordingly
when it is constructed.

In case there is no link do not attempt to submit a packet to the packet
stream but return with an error so that upper la [... cut]
closed 2016-11-30 by chelmuth