genodelabs/genode #2119: hw: remove usage of deprecated printf

skalk, 2016-10-07:
According to issue #1987 we shall remove all `printf` calls from base-hw specific code.

m-stein, 2016-10-07:
I would prefer to replace the calls with an alternative and, AFAIK, there is currently no such alternative. (log(), warning(), error(), and raw() do not work in the kernel)

skalk, 2016-10-09:
@m-stein of course they do, as far as we initialize the log environment (grep for init_log).

skalk, 2016-10-09:
When removing printf from base-hw's core implementation completely, one gets linker errors because of some usage of printf within Allocator_avl_base::dump_addr_tree and a special case  in Log_session_component::write (file repos/base/src/core/include/log_session_component.h:70). Whereby the first occurrence is easily replaced, the second is used to cut a linebreak from a line that only contains an escape sequence ending. Because it tries to prevent a linebreak, it cannot be easily replaced with `log` which always ends with a linebreak. @nfeske am I right? Do you see an easy solution to get rid of those last printf usage? From my point of view, I'm wondering whether we need this special treatment at all?

nfeske, 2016-10-10:
The best way would certainly be the use of the new `Buffered_output` utility (base/src/include/base/internal/output.h) in the log-session implementation. This way, the logic for batching and flushing subsequent write request will be implemented at only one place. I cannot fully remember the rationale behind the special escape-sequence handling. I guess that the heuristic tries to reduce the chance that characters are flushed in the middle of an escape sequence? But since I am not 100% sure, I am fine with dropping the special treatment.

skalk, 2016-10-10:
To really get rid of printf we have to remove it from generic core code too, so I extend this issue to all base-* repositories

commit 46d27219 (2016-10-18): base-*: remove usage of printf

base generic code:
  * Remove unused verbosity code from mmio framework
  * Remove escape sequence end heuristic from LOG
  * replace Core_console with Core_log (no format specifiers)
  * move test/printf to test/log
  * remove `printf()` tests from the log test
  * check for exact match of the log test output
base-fiasco:
  * remove unused Fiasco::print_l4_threadid function
base-nova:
  * remove unused hexdump utility from core
base-hw:
  * remove unused Kernel::Thread::_print_* debug utilities
  * always print resource summary of core during startup
  * remove [... cut]
closed 2016-10-21 by chelmuth