genodelabs/genode #2120: Remove upcallls to root interface

nfeske, 2016-10-07:
In the current version of Genode, the parent of a service calls the `Root::session`, `Root::close`, and `Root::upgrade` RPC functions of the server child to manage sessions. Because the parent (as the client of the root interface) can never be sure that the untrusted child actually responds to these requests, the init component issues session-management requests from different threads.

It would be preferable to eliminate the need for one thread per init child, for the following reasons:
* The maximum number of threads per component is limited by the stack area. This imposes a hard limit on the maximum number of children of init.
* Threads consume resources that must be accounted.
* The threads within init operate on shared data structures, which implies the need for synchronization.
* If a child blocks in a root-RPC call, it cannot be easily destroyed, which prevents a solid implementation of a dynamically reconfigurable init, where children must be destructible regardless of their behavior.

For these reasons, both use cases of the current parent and root interfaces must be covered by an asynchronous interaction between the parent and children.

Still, we need to uphold API compatibility to the current synchronous root interface - at least for a while. I have the following idea to achieve this:

* We replace the `Parent::announce` RPC function by a component-locally implemented class function with the same name. When called for the first time, it creates a new "root" entrypoint (ep) with the sole role to handle session-management requests. The new entrypoint requests a ROM module "session_requests" from its parent (i.e., "init"). This ROM module is populated by the parent with pending session creation/close/upgrade requests in the form of XML nodes. New requests are signaled via the existing ROM-update mechanism.

* When the `Parent::announce` function is called, it records the association of the provided service name and corresponding root capability in a local data structure (service registry).

* Once the root ep receives an update of the "session_requests", it goes through the XML nodes of the current requests and issues local RPC requests to the root capability as found in the service registry. This call is synchronous but it stays within the component. Each session present in the "session_requests" ROM has a parent-defined session ID. The ID is solely used between the parent and the server child.

* If the component-local root component returns a session capability, the root entrypoint calls a new `Parent::deliver_session(session_id, cap)` RPC function of the parent interface, which delegates the session capability to the parent.

* In response the close and upgrade requests, the root entrypoint calls the new `Parent::close_ack` and `Parent::upgrade_ack` RPC functions with the session ID as argument.

The root entrypoint should be seen as an intermediate mechanism. In the future, components will be able to use the asynchronous interface directly and thereby eliminate the need for the root entrypoint. We will certainly come up with a convenient equivalent of the `Root::Component` class that directly operates of the asynchronous interface (e.g., called "session factory"). Once all servers will be converted to the session-factory mechanism, we will be able to drop the entire notion of the original root interface.

m-stein, 2016-11-14:
Another fixup for the CPU sampler.

m-stein, 2016-11-14:
The errors with CPU-sampler noux should get fixed too by the CPU sampler fixup.

m-stein, 2016-11-14:
Three commits that fix the thread test. Commit https://github.com/m-stein/genode/commit/aa96da8da5945b73cc04a62a98b72300dd2b059a fixes a problem that might exist also in other session-component calls that create RPC objects for the client.

cproc, 2016-11-14:
Fixup commit 6704eee29520113cd64a9e73bdc2ad522b5567c7 for qt5.

m-stein, 2016-11-21:
The new fixup should fix the timer test on hw_x86_64.

m-stein, 2016-11-21:
Updated the fixup to use internal_env().

nfeske, 2016-11-21:
@m-stein Thank you for the new commit - very nice!

m-stein, 2016-11-21:
I've added another commit on my branch (async_parent_root_interface). This commit is an addition to https://github.com/genodelabs/genode/commit/10939814276b3a15f4b45c030c1eb1a1c1b5ca13 and references the other cap-ref-count overflows in core. It also adds a test that attempts to trigger some of them. Unfortunately, this test can't trigger all of them on each platform. For example, most platforms don't provide enough IRQs to trigger the overflow in Irq_session::sigh (this is why I don't test it at all). Or, some kernels don't let me create enough threads.

m-stein, 2016-11-22:
The new fixup should fix the problems with hw_x86_64 since my previous fixup ea70f92dcfd5e1d255684688172edcbb8f4abc06.

cproc, 2016-11-22:
Fixup commits for Arora, qt5_avplay and gdb_monitor.

@nfeske, @chelmuth: please add to staging

nfeske, 2016-11-23:
Thanks @cproc for the fixups - merged to staging.

cproc, 2016-11-24:
Fixup commit d9d3a304182ec38a43d197c6615cf2a6b2145a39 fixes the GDB monitor compile error for 32-Bit platforms.

@nfeske: please add to staging

nfeske, 2016-11-24:
@cproc thanks, added.

cproc, 2016-11-28:
Fixup commit f69a442a8cbba6c9c0b3b056850735509cc30367 adapts the qt5_avplay framebuffer filter feature to the new API. The changes in the framebuffer session header files were necessary for using the 'Single_session_service' class with the framebuffer service.

@nfeske, @chelmuth: please add to staging