genodelabs/genode #2125: Log debbugging

ehmry, 2016-10-11:
I like having a log debugging function around that is easy to hunt down during code cleanup.

May it never be used in the master branch : -)

ehmry, 2016-10-11:
BTW, the color here is cyan.

nfeske, 2016-10-11:
I would prefer moving the `PDGB` macro from printf.h over to log.h, e.g. like this (I have not tested the snippet):

```
#define PDBG(...) \
	Genode::log("\033[33m", __PRETTY_FUNCTION__, "\033[0m", ##__VA_ARGS__);
```

It is simpler and we would always get the messages prefixed with the name of the calling function. Would that work for you, too?

ehmry, 2016-10-11:
I don't like implicitly using __PRETTY_FUNCTION__ because it can get very noisy in the case of lambdas.

ehmry, 2016-10-11:
Also, I like using a proper namespaced function or a separate header because it makes it easy to turn debug lines into compile errors.

nfeske, 2016-10-12:
You are right that using a separate header is more sensible than adding such debug utilities to log.h.

The use of PRETTY_FUNCTION is our time-tested approach, which I'd like to keep.

I think that the use of debug macros varies quite a lot from developer to developer. I tend to sporadically insert a few PDBGs while debugging. Once I have validated my assumption, I usually remove them immediately. When committing, I grep for `PDBG` (which is a very nice term to grep for because nothing else is named so awkwardly) and remove remaining occurrences. In practice I haven't suffered under the noise of PRETTY_FUNCTION too much. But I haven't used PDBGs within lambdas much either.

I want to prevent the further proliferation of rainbow-like log output (at least as produced by the code that is checked in). Intuitively, the addition of cyan goes against my grain. However, when offering the feature only when including debug.h and enforcing the rule that debug.h must never be included in any checked-in code, I think I can live with it. As a compromise, may we add the `PDBG` macro in addition to your `debug` function in debug.h, but settle for the yellow color that we currently use for `PDBG`? ;-)

ehmry, 2016-10-13:
I'm happy with just a macro in a separate header. Keeping the name and the output format the same makes debug lines look distinct in the code and in the logs. And yellow is fine with me now that PLOG is going away.

Redefining PDBG is going to break in a few places where there are types that aren't recognized by Output, but if the debug header isn't in use for checked in code then it shouldn't be a problem.

nfeske, 2016-10-18:
Thanks for the new commit!

ehmry, 2016-10-18:
now Sans-semicolon

commit ef5f36bf (2016-10-21): Redefine PDBG to use new log output function

PDBG is redefined in base/debug.h, an aide to development which is not
to be included in release code.

Fix #2125
closed 2016-11-08 by chelmuth