genodelabs/genode #2138: vfs: strange hang when filesystem is full

skalk, 2016-10-18:
I stumbled across this problem when using the Heeselicht scenario. When copying into a filesystem, in this case ext2 with rump_fs, where there is not enough space left, the copy operation of the bash never stops, and produces a lot of garbage escape sequences on the terminal.

skalk, 2016-10-18:
above commit contains a more workable run-script than Heeselicht to reproduce the issue.

chelmuth, 2016-10-18:
@cnuke IIRC you have additional information regarding this topic.

cnuke, 2016-10-18:
Yes, @skalk already tried the quick-and-dirty fix-attempt (*written > 0 ? OK : ERROR*) but it did not help. So we have to investigate further.

nfeske, 2021-02-26:
I'm closing this issue because it has become stale.

closed 2021-02-26 by nfeske