genodelabs/genode #2139: nic_router: more intuitive config interface

m-stein, 2016-10-18:
During some offline discussions with different potential users of the NIC router, each having his/her own use case and requirements, the demand for a new, more intuitive configuration scheme for the NIC router emerged. This is an example for the new scheme:
```
<policy label="uplink" interface="10.0.2.55/24" gateway="10.0.2.1">
	<tcp-forward port="80" label="http_srv" to="192.168.1.18" />
	<ip dst="10.0.98.0/24" label="udp_cln" />
	<nat label="http_udp_cln" tcp-ports="6" udp-ports="5" />
	<nat label="udp_cln" udp-ports="3" />
</policy>
<policy label="udp_cln" interface="10.0.98.33/24" gateway="10.0.98.33">
	<udp-forward port="1337" label="udp_srv" to="10.0.99.55" />
	<ip dst="10.0.2.0/24" label="uplink" />
</policy>
<policy label="http_udp_cln" interface="100.200.0.1/24" gateway="100.200.0.1">
	<tcp dst="10.0.0.0/16">
		<permit port="8080" label="uplink" />
		<permit port="80"   label="uplink" />
	</tcp>
	<udp dst="10.1.0.0/16"> <permit-any label="uplink" /> </udp>
</policy>
<policy label="http_srv" interface="192.168.1.1/24" gateway="192.168.1.1" />
<policy label="udp_srv"  interface="10.0.99.33/24"  gateway="10.0.99.33"  />
```
Each interface has its own subnet. To enable subnets to span over multiple NIC sessions is not a task of the NIC router. The subnet of an interface as well as the NIC router identity in this subnet is given through the "interface" attribute in a policy. The gateway attribute of a policy replaces the former "via" attributes of routing rules. It tells the NIC router which next hop to use for a packet that shall be send to this interface but whose IP destination doesn't match the subnet. For TCP/UDP port forwarding, there are now dedicated rules with mandatory "to" attribute. They apply only if the incoming packet addresses the NIC router identity in the senders subnet.

The "tcp" and "udp" rules are also new. They apply if the incoming packet contains the same transport protocol, its IP destination matches the "dst" attribute, and there is a permit rule for its destination port. A "permit-any" rule overrules all specific "permit" rules. All port forwarding, TCP and UDP rules create a link state when applied. This link state references the corresponding TCP/UDP connection for future packets not only from the client but also from the server side. On incoming packets, the router always searches for a matching link state first before using any static rules. This is why "http_srv" and "udp_srv" in the example do not need any routing rules. 

In contrast to that, "ip" rules simply redirect packets if the "dst" attribute matches their IP destination. This is why "uplink" and "udp_cln" in the example both have an IP rule to each other to allow for bidirectional communication.

The NAT configuration has moved from its former place in the "source" policy to the "destination" policy. A "nat" rule now tells the router to NAT the interface named by the rule ("label" attribute) if it sends to the interface whose policy contains the "nat" rule. The attributes "tcp-ports" and "udp-ports" now refer to the interface-local port space. So effectively the NIC router now has a complete port space available for NAT at each interface and not only one for all, like beforehand.

m-stein, 2016-10-18:
I have the NIC router now in a state where it speaks the configuration defined above. I was able to remove some unnecessary overhead especially in the code paths that search for appropriate link states or rules. I also prepared the router for the introduction of runtime configuration changes and replaced the use of pointers by references and exceptions as far as sensible. Currently I'm cleaning it all up.

m-stein, 2016-10-19:
I've made the "gateway" attribute optional as it is not useful at subnets where the NIC router is the default gateway (normally the case for all downlinks). If the "gateway" attribute is not given for an interface, subnet-foreign packets to this interface are dropped with an error message.

m-stein, 2016-10-24:
The new version of the NIC router is ready so far except for the README and the commit message of the final commit. I've uploaded it all so it may be reviewed already. Next I'll write the README and the missing commit message.

m-stein, 2016-10-27:
I've updated the branch (nic_router_enhanced). I've added the README and fixed some stuff that I found while writing. Now, TCP links have a general timeout of 2 times the round trip time, a pagefault on session destruction was fixed, and in the nic_router.run script I renamed srv server and clnt client.

m-stein, 2016-11-02:
I had some spare time this evening and used it for moving the NIC router to the new Component API. This implied replacing the net/dump header by print methods at the different packet types. The two commits are now at my branch (nic_router_enhanced).

m-stein, 2016-11-03:
@chelmuth Thanks for your hints. I've updated the branch (nic_router_enhanced) accordingly.

m-stein, 2016-11-24:
@chelmuth @skalk May we merge this to staging?

chelmuth, 2016-11-24:
Beside the notes I have no objections to merge the branch.

m-stein, 2016-11-24:
@nfeske @chelmuth Thank you for your review and feedback! I have adapted my branch (2139_nic_router_more_intuitive_config_interface) accordingly and also applied the new timeout framework. This is why these commits now depend on the commits of #2170.

m-stein, 2016-11-25:
@chelmuth I added differentation of policy and domain tag, and adapted the test on my branch (2139_nic_router_more_intuitive_config_interface). A README update is missing but I'll add it during the weekend.

chelmuth, 2016-11-25:
I have two tiny remarks. First, a test of `run/nic_router` on foc_x86_32 just timed out. Second, I'm wondering why the test (incl. http_clnt) should not run on other platforms beside foc. @m-stein what do you think? Nevertheless, current staging reflects your current state of development.

m-stein, 2016-11-28:
@chelmuth The new commits make it work on x86-64, prevent multiple sessions at one policy, and update the README. I'll continue working at x86-32 as soon as I have my contribution for the release notes ready.

chelmuth, 2016-11-28:
Merged your commit to staging (current is ec1379b3cc776cf981e9158ebfe3a39c797d7a1d). Please consider to rebase further work on top.

commit 25b38113 (2016-11-25): nic_router: new user interface and optimizations

Fixes #2139
closed 2016-11-30 by chelmuth