genodelabs/genode #2150: Allow RPC arguments w/o default constructor

nfeske, 2016-10-26:
@3dik posted the following suggestion to the mailing list (https://sourceforge.net/p/genode/mailman/message/35449120/):

```
Hi,
as described in "Genode Foundations", all types which are used as  
argument or return types in RPC functions must have a default  
constructor. I do not think that such a limitation is really  
necessary. So I tried to write a proof of concept [1]. These  
modifications should allow RPC functions to pass most of the  
non-default-constructible objects which satisfy the rest of the  
requirements for RPC argument and return types. I extended the  
"hello_tutorial" with a little test code to show that it works [2].

Software patterns like RAII [3] are a possible reason for removing  
that limitation. Quite often classes cannot provide a useful  
default-constructor which does more than merely setting the object  
state to "invalid". Because often there are no useful default values.  
So if a class provides a default-constructor which just sets the  
object to "invalid", the code complexity increases since care has to  
be taken that the object is fed with data up to a specific point of  
time.

Unfortunately, my implementation does not remove default-constructor  
calls completely:
1. Capabilities still need default-constructors. Msgbuf stores them  
directly in an array and I did not find the time to check changes of  
this storage behavior for possible side effects.
2. Output-only _arguments_ have to be default-constructible too.  
Please note, that non-default-constructible _return_ types are  
allowed. So this affects only RPC arguments which are explicitly  
declared as Output-only, for example by adding custom specializations  
of Rpc_direction. The reason for this limitation is that my Pod_tuple  
version still stores the RPC arguments directly. While input arguments  
are copy-constructed, it is hard to copy from an object that does not  
yet exist.

For the second point, I see no simple solutions: we could create a  
simple byte buffer instead of a default-constructed output-only  
argument. Since "normal" RPC arguments have to be trivially copyable  
[4], a RPC function could easily write to this buffer, which would be  
accessible by the correctly type-casted reference or pointer. A  
minimalist version of boost::optional [5] or the like as member of  
Server_args could make sense. However, it would require more template  
specializations and case differentiations because the special RPC  
types Capability and Native_capability are not trivially copyable.

I successfully tested the code with the repositories "hello_tutorial"  
and "demo" on base-linux. But if you decide to use my code  
contribution, please note that it is rather a proof of concept. There  
is room for improvements:
* I tried to minimize the code modifications. So some refactoring may  
be useful: I expanded Meta::Empty by constructors and a typedef, I  
inserted possibly confusing forward declarations, and Meta::Pod_tuples  
and Meta::Pod_args could be renamed because they do not only accept  
POD types [6].
* I completely ignored the trace-related code. I do not know if there  
is a need of modifications.

Thanks to Johannes Schlatow who gave me the idea of grappling with this topic.

[1] https://github.com/3dik/genode/tree/nodef
[2] https://github.com/3dik/genode/commit/c35a8d4a3166da74c46423f5e0b4a35933174f3d
[3] https://en.wikipedia.org/wiki/Resource_Acquisition_Is_Initialization
[4] http://en.cppreference.com/w/cpp/concept/TriviallyCopyable
[5] http://www.boost.org/doc/libs/1_62_0/libs/optional/doc/html/boost_optional/tutorial/when_to_use_optional.html
[6] http://en.cppreference.com/w/cpp/concept/PODType
```

3dik, 2016-10-27:
>> Unfortunately, my implementation does not remove default-constructor calls completely:
>> 1. Capabilities still need default-constructors. Msgbuf stores them directly in an array and I did not find the time to check changes of this storage behavior for possible side effects.

> This is not a problem because a default-constructed 'Capability' has a well-defined 'invalid' state, which is expected (like for the boost::optional types you cited).

This is true but the boost::optional types would be used only internally in the RPC code, which in turn would narrow down the above-mentioned increase of code complexity. In contrast to that, 'Capability' objects may be constructed outside of the RPC code and a method call on an invalid 'Capability' object can lead to runtime errors.

So if the RPC mechanism was the only reason for allowing such a 'invalid state', it _could_ be worth redesigning the 'Capability' and 'Msgbuf' classes. Of course, the current design is not a big problem and I cannot assess the effort of a redesign. I just want to make clear that it would actually _be_ at least a small improvement if there is no "real" reason behind the default-constructibility :)

>> I inserted possibly confusing forward declarations, and Meta::Pod_tuples and Meta::Pod_args could be renamed because they do not only accept POD types [6].

> I agree. With your changes, the (renamed) template should be moved outside of 'meta.h', preferably to 'base/rpc.h'.

Another approach came into my mind :-D. Instead of calling constructors recursively, we could make use of [aggregate initializations](http://en.cppreference.com/w/cpp/language/aggregate_initialization). This way, we keep Pod_tuple clean, for example by defining a new recursive _read_args method in `rpc_server.h.` An example ( I switched the argument order for simplicity ):

- construct an Meta::Empty object 'e' and extract the first argument 'first' of type T1
- construct `Pod_tuple<T1,Empty>` with the braced-init-list { first, e }, name it 'previous_tuple'
- extract the second argument 'second' of type T2
- construct `Pod_tuple<T2,Pod_tuple<T1,Empty> >` with the braced-init-list { second, previous_tuple }
- and so on

ValiValpas, 2016-10-27:
I'm impressed ;-)

nfeske, 2016-11-01:
> So if the RPC mechanism was the only reason for allowing such a 'invalid state', it could be worth redesigning the 'Capability' and 'Msgbuf' classes.

The invalid state of a capability is actually not a result of the RPC limitations. We deliberately use it, e.g., to uninstall a signal handler, we install an invalid signal-context capability. Usually however, the receiver of a capability originating from a server expects the obtained capability to be valid. It is not needed to explicitly check it before using it. A check would be futile anyway because a (misbehaving) server may stop responding to RPC calls at any time still. In principle, I like your idea, but I would not pursue it right now as it does not solve a pressing problem.

Just as side note, we actually have something similar to `boost::optional`, albeit motivated by a different use case: https://github.com/genodelabs/genode/blob/master/repos/base/include/util/volatile_object.h (admittedly, it is named a bit strangely but we plan to rename it in issue #2151)

It goes without saying that I would prefer `Pod_tuple` to stay generic instead of moving it to rpc.h. Are you going to try out your idea? If yes, I would defer merging your work until you are done.

3dik, 2016-11-03:
> Usually however, the receiver of a capability originating from a server expects the obtained capability to be valid.

And this is completely okay. But RPC interfaces such as 'Rm_session' can provide methods which send a 'Capability' as return value to the client. A programmer could construct an invalid Capability object and forget to setup it with a call of the correct RPC method, which is 'create' in the Rm_session example. If the Capability class was not default-constructible, this mistake would immediately lead to a compilation error instead of a runtime error. So the only additional check would be a compile time check of the existence of a default constructor.

There are other similar scenarios. However, as you mentioned, the rarity of such scenarios and the use cases of the invalid state limit the advantages of such a redesign.

> Are you going to try out your idea? If yes, I would defer merging your work until you are done.

Well, the commit history might be confusing. I was unsure whether I should push my changes to the old branch or create a new one. I chose the latter, replaced the argument-related commit ( [old](https://github.com/3dik/genode/commit/1f27eb8702005f4b293b6ae6b05a594ac18431a0), [new](https://github.com/3dik/genode/commit/2708d9405f92fe06ab47569e7d65f0130f8f904e) ) and reordered slightly the history. The new branch is "[nodef-aggr](https://github.com/3dik/genode/tree/nodef-aggr)"

Some notes:
- We could simplify the code by completely removing the support of output-only arguments.

- - A super-rare case: RPC methods with output-only arguments may lead to compiler warnings ( ... be used uninitialized in this function ). If you want to support not only 'Capability' as output-only arguments, you could use the class 'Return', which I removed in the [return-value-related commit](https://github.com/3dik/genode/commit/e161e73f2e049f509322cb37c002ccc9f113c185#diff-9adb2390f2f893a6a29202cf3f0314aaL140).
- [The last commit](https://github.com/3dik/genode/commit/771382834dbc78ad7a03349f44bd267775818fbb) is not directly related to his ticket. But it's small :-D.

nfeske, 2016-11-08:
> A programmer could construct an invalid Capability object and forget to setup it with a call of the correct RPC method, which is 'create' in the Rm_session example.

Now I understand. Thanks for the explanation. In fact, I ran into this problem several times and would have appreciated a compile error in these situations. Also, if an invalid capability could be constructible solely via a `Capability::invalid()` class function (which may use the private default constructor), the deliberate creation of an invalid capability would become much more clear. Right now, the code that constructs such capabilities looks far innocent.

Thanks for taking the time for cleaning up the history and for implementing the aggregate initializations idea - very cool! I reviewed your patches - they look very good! Tomorrow, I will go though them together with @chelmuth and likely merge the current patch series to staging to let our automated test chew on it.

> We could simplify the code by completely removing the support of output-only arguments.

I won't hold you back. The less complex the code becomes, the better! ;-)

chelmuth, 2016-11-10:
From my side, the patch series (except the test in hello) can be merged into staging if the style fixes in 4c81181 are confirmed. I propose to squash the series into one commit with a detailed commit message that aggregates all information from the current individual commits. Thanks @3dik for the great contribution.

3dik, 2016-11-19:
> Also, if an invalid capability could be constructible solely via a Capability::invalid() class function (which may use the private default constructor), the deliberate creation of an invalid capability would become much more clear.

This is also an option of course.

> ... if the style fixes in 4c81181 are confirmed

If you have referred to me, I confirm this of course. To be honest, I cannot remember why I inserted all the spaces. However, an explicit rule in the style guidelines might be useful.

chelmuth, 2016-11-25:
I'm planning to merge ccc0526b27230b2272cb26e30e0cd005509994a2 and ea6c5c74d1f48304715fa96b6bbd169862d41b27 today because commits in #2170 can directly make use of them. Any objections?

nfeske, 2016-11-25:
I agree.

commit cb3cf535 (2016-11-25): Allow RPC arguments w/o default constructor

Fixes #2150
commit a285b940 (2016-11-25): Allow RPC arguments w/o default constructor

Fixes #2150
commit 59e5b392 (2016-11-28): Fixup "Allow RPC arguments w/o default constructor"

Fixes #2150
closed 2016-11-30 by chelmuth