genodelabs/genode #2151: Give `Lazy_volatile_object` and `Volatile_object` non-awkward names

nfeske, 2016-10-26:
When I originally created the utilities in `util/volatile_object.h` during my work on nitpicker, I did not foresee that those utilities would be _that_ generally useful. Back then, I settled on the names because they express that the enclosed object may not stay the same over time but I was never happy about the naming.

Now that those utilities are used everywhere, I'd prefer to use more concise names. I suggest to rename `Lazy_volatile_object` to `Constructible` and `Volatile_object` to `Reconstructible`. The header should be renamed accordingly to `reconstructible.h`. The rationale behind the new names is as follows:

`Constructible` expresses that the enclosed object can potentially be constructed but not definitely and not immediately.

`Reconstructible` expresses that the enclosed object is just like a normal object, but with the added feature of its constructor being callable multiple times (with potentially different arguments). Of course, a `Constructible` object is implicitly reconstructible.

Are there any other opinions?

chelmuth, 2016-10-26:
As discussed offline I support the renaming.

m-stein, 2016-10-27:
I like the idea of renaming them, especially because the new names are less bulky. However, to me, Constructible seems to be a little confusing as every C++ object is constructible, but admittedly, I have no better suggestion.

commit 25a7ea3d (2016-12-01): base: rename 'Volatile_object' to 'Reconstructible'

Fixes #2151
closed 2016-12-14 by nfeske