genodelabs/genode #2154: vfs: block_file_system client crashes because of unhandled Packet_alloc_failed

alex-ab, 2016-10-27:
Seen with vbox using ahci model. The exception is not handled here:

https://github.com/genodelabs/genode/blob/master/repos/os/include/vfs/block_file_system.h#L77

alex-ab, 2016-10-27:
Does this commit qualify as stopgap fix ?

chelmuth, 2016-11-03:
@ehmry I have the feeling this will be silently fixed with the new async VFS?

ehmry, 2016-11-04:
It should, but so far I have neglected to write a proper async plugin.

alex-ab, 2016-11-04:
I have now the confirmation that the two commits fixes the issue of a customer. If the commits go not online, which I can understand, please make sure that the Packet_alloc_failed case during alloc_packet is handled properly.

ehmry, 2016-11-04:
If there is a problem I say push the fixes until the new block client is ready.

alex-ab, 2016-11-28:
@chelmuth, @nfeske: please consider the commit for staging, if the async vfs does not make it to the release

chelmuth, 2016-11-28:
I merged 16d55d1f83a7fb711aa9b4361736549eeb260f2c. @alex-ab please check if I resolved the merge conflicts correctly.

alex-ab, 2016-11-28:
Looks good, thx

commit de81e852 (2016-11-28): vfs: handle alloc failed in block_file_system

Fixes #2154
commit 16d55d1f (2016-11-28): vfs: handle alloc failed in block_file_system

Fixes #2154
closed 2016-11-30 by chelmuth