genodelabs/genode #2157: run tool support for Genode hosted Linux development

ehmry, 2016-10-31:
I would like to pass a run scenario from Linux in Virtualbox down to a Genode host, my motivation is network testing with full routing.

It should be possible to copy binaries and configuration thru a shared folder, and start a execute a the scenario by triggering a configuration reload in a running init instance. I will find out.

ehmry, 2016-11-01:
On second thought, better to control an instance of cli_monitor through a terminal file.

nfeske, 2016-12-01:
Would you like to keep the issue open or is it obsolete?

ehmry, 2016-12-01:
I would like to keep it open, I want to get ROM support and notifications into the VFS server, and then that should handle all the necissary configuration pushing and updating.

ehmry, 2016-12-06:
No, I close this and reopen later.

closed 2016-12-06 by ehmry