genodelabs/genode #2159: base: iterator for AVL tree

skalk, 2016-11-02:
While implmenting issue #2092, again I hit the problem to iterate over an AVL tree. I think we need this very basic functionality several times, so I would like to integrate it into the AVL tree itself. Thereby we can also replace the current debug print utilities within Allocator_avl.

m-stein, 2016-11-02:
@skalk I've also needed such an iterator for the NIC routrer and implemented it in  https://github.com/m-stein/genode/commit/2dc1901ea79291b3432b466fe2fc59ec84230abd

m-stein, 2016-11-02:
@skalk Your code implements an in-order traversal while mine implements a post-order traversal. For my use case it isn't relevant in which order the tree is traversed but I would suggest to name the iterator according to the traversal type in general.

skalk, 2016-11-02:
Ok, I've added a comment, changed the template type's name, and reworked Rom_fs to use the new 'for_each'. @nfeske: would you mind to review it once again?

skalk, 2016-11-02:
No problem! I will change it.

skalk, 2016-11-02:
@nfeske ok, finally done.

commit 1542d921 (2016-11-08): base: iterator for AVL tree (fix #2159)
closed 2016-11-30 by chelmuth