genodelabs/genode #2160: VFS auditing

ehmry, 2016-11-02:
I made a little amendment to the VFS that has come in handy in a few different situations.

When the top level ``<vfs>`` node or any ``<dir>`` nodes have ``audit="yes"`` attached to them a log of file accesses is printed to a dedicated log session.

This is handy for debugging programs, for instance, does it open /etc/resolv.conf? What at the search locations for data files? Is the VFS working at all?

nfeske, 2016-11-02:
Sounds handy. But I wonder, couldn't this be implemented as a VFS plugin? If we pass the root directory to the FS instances, the audit FS could work as follows:

```
<vfs>
  <dir name="flaky">
    ... something strange happens here
  </dir>
  <audit name="audit" path="/flaky"/>
</vfs>
<policy label="client" root="/audit"/>
```

In a similar fashion, different audit-like VFS plugins may provide different statistics (e.g., performance measurements). Granted, it may by slightly less convenient to use in your particular scenario. But the approach would scale better while keeping the VFS untouched.

ehmry, 2016-11-02:
Yes, a plugin would be cleaner, when I get the socket stuff out of the way I will try get the root directory argument in place.

ehmry, 2017-08-03:
I now see this as related to #2467. Information from running raw performance benchmarks is useful but should be combined with less contrived data. Collecting a file-system "trace" of a scenario like ``noux_tool_chain_auto`` and replaying only the file-system transactions would be one way of quickly gathering performance data over different VFS configurations.

ehmry, 2018-03-22:
Fixed on my **issue2160_vfs_trace** with a VFS plugin that is constructed with the root directory of the VFS. Requires #2701.

I named this plugin 'audit' and not 'trace' because it doesn't track I/O operations. I think that is part of a solution to a problem separate from just trying to figure out what an application is trying to do.

nfeske, 2018-03-22:
I take you commit as confirmation that my root-dir commit works as intended? That's nice to hear. ;-)

Naming the plugin "audit" is a very good idea.

Do you see a way to get rid of the `EXPAND` macro?

ehmry, 2018-03-22:
Well the `EXPAND` macro made the plugin faster to write but it could be replaced with an explicit stack `Path` in every method.

ehmry, 2018-03-22:
Or the VFS API could be changed to take paths in a different format than null-terminated strings ;-)

nfeske, 2018-03-23:
I wonder, why wouldn't a private method work just as well? Somehow along these lines:

```
Absolute_path _expanded_path(char const *path) const
{
  return Absolute_path(path + 1, _audit_path.string()).string();
}
```

> Or the VFS API could be changed ...

Yeah. I agree with this statement (generalization be me). Fortunately, the VFS API is not exposed to regular component developers. So we can still move rather freely. There are still several legacies of the past that I like to see removed such as the ioctrl stuff, or use of out parameters. That said, we should leave this cleanup topic for another day once the current line of feature development has settled.

ehmry, 2018-03-23:
Fixup at 2119a2b40951145ad38902c71f2b3b143cb3967c to remove macro.

nfeske, 2018-03-23:
The commit 2119a2b40951145ad38902c71f2b3b143cb3967c changes the framebuffer session but not the VFS audit plugin.

ehmry, 2018-03-23:
Woops

ehmry, 2018-03-23:
Fixup at 0509b2213184e65dfd4a0e96ed85df9d6729ad8b.

ehmry, 2018-03-27:
This plugin should be built into the VFS library or moved to `repos/gems/src/lib/vfs`. Otherwise it becomes part of the VFS API package when `repos/os/src/lib/vfs` is copied to the depot.

nfeske, 2018-03-30:
@emery Would you mind adjusting your commit to the new location at gems and rebasing it  to staging? I guess that it needs a slight adaptation to the VFS-root-dir change. Please feel free to create a new commit instead of a fixup.

ehmry, 2018-03-31:
No problem, I've rebased the whole series on my **issue2160_vfs_audit** branch.

emery, 2018-03-31:
Hello,
I am not the right 'emery' you're looking for.
cheers

nfeske, 2018-04-01:
@emery thanks for the notice, and sorry for the confusion. ;-)

@ehmry thank you for the quick rebase! I added both commits to staging.

commit c0c1cd1e (2018-04-01): VFS audit plugin

Plugin for auditing VFS access using the VFS server. Useful for tracking
which files ported software expects to be present.

Fix #2160
closed 2018-04-10 by chelmuth