genodelabs/genode #2165: Remove Connection::KEEP_OPEN feature

nfeske, 2016-11-06:
This feature allows a connection object to be temporarily created while keeping the session open even after the connection object is gone. With the forthcoming redesign of the parent interface to become non-blocking (#2120), we have to drop this feature because the client-side session state will entail more than just the session capability, in particular it comprises a session ID, which is naturally stored in a connection object and should be deallocated not before the session is closed.

In hindsight, KEEP_OPEN was a misguided convenience feature. It was introduced to accommodate situations where a connection object could not be easily held as an object member because its initialization required several procedural steps. Today, we have the `Lazy_volatile_object` utility to deal with such situations. So we can happily live without the `Connection::KEEP_OPEN` feature.

nfeske, 2016-11-06:
The commit above removes the use of Connection::KEEP_OPEN from all places except for Qt5. @cproc Could you please take care of adjusting libports/src/lib/qt5/patches/qt5_qtbase_genode.patch accordingly? @alex-ab Could you please confirm that my changes of the ACPI driver are ok?

alex-ab, 2016-11-07:
@nfeske: beside my two comments it looks good - if it still works

nfeske, 2016-11-07:
@alex-ab Thanks for the review. I added the fixup above.

cproc, 2016-11-07:
The Qt5 part is done in issue #1631.

nfeske, 2016-12-01:
Merged into master.

closed 2016-12-01 by nfeske