genodelabs/genode #2166: init: print out reason for session creation error

trimpim, 2016-11-07:
We have a relative complex scenario with many components and much more routes between them.
It is no trivial task to find out why the creation of a session failed without specific error message.
A short message should be printed so that the different cases can be distinguished.

nfeske, 2016-11-07:
I don't think that the low-level child.cc code is the appropriate place for these messages. From the child's perspective, this condition is an error. But from the parent's perspective (where the child.cc code resides), this condition does not present a problem.

That said, I agree that printing the diagnostic information is valuable. How do you think of adding the warning to the `Connection` class instead. Please have a look at my commit e63bfab074323a46d8064ca245273ec1c582f796

trimpim, 2016-11-08:
In  principle I'm not against moving the information to a better location. But it should still be possible to distinguish between the different cases, otherwise we will end up with the same situation we have now, that we need to dig in to the code (add error messages) to find out what went wrong.
IMHO it is an error if the connection can't be opened.

ehmry, 2016-11-08:
I think some guidelines on when and where errors should be printed to the log would be helpful.

nfeske, 2016-11-08:
I am actually planning to remove the distinction between the `Service_denied` and `Unavailable` exceptions of the `Parent` interface because it should be of no matter to the child to know whether the requested service exists or not. The message that a connection of a certain type could not be established is certainly useful. This is covered by my patch. Furthermore, if init is not able to route the session to a service, it actually prints a message. E.g., try to change the following routing configuration in hello_tutorial/run/hello.run:

```
  <start name="hello_client">
    <resource name="RAM" quantum="1M"/>
    <route>
      <any-service> <parent/> </any-service>
    </route>
  </start>
```

With this configuration, there exists no route to the hello server. When starting the scenario, init prints the following message:

```
[init] Warning: hello_client: no route to service "Hello"
```

Hence, for the `Unavailable` condition, init already prints the diagnostic information. For the `Service_denied` condition, my patch causes the client to print the information:

```
[init -> hello_client] Warning: Hello-session creation failed (ram_quota=4K)
```

Regarding the distinction between warnings and errors, this is a matter of perspective. What may be catastrophic for one component (e.g., hello client aborts because it cannot open a session to the hello server) is a mild inconvenience for another (init noticing that one child is apparently doing "stupid" things like requesting a session it shouldn't access). So maybe it would be sensible to print the message in the `Connection` object as an error?

Please consider the following remark when comparing your suggestion to modify the base code with my suggestion to let higher-level code take are of the messages. Adding such messages to the base code effectively prevents components from suppressing the messages. In the worst case, a misbehaving child may deliberately spam the log of its parent by repeatedly requesting sessions to unavailable servers. If the messages are printed by init (like the routing error), init or a custom runtime environment could take precautions like limiting the rate of messages of the child. But by having the messages always in the base code, this is not possible.

trimpim, 2016-11-09:
After reading your explanation, I see, why your solution is a better solution, than the one we had before.

Regarding warning vs. error, for me personal this is not that important, as long as there is  a diagnostic message of any kind. But some of my co workers see this different.

nfeske, 2016-11-09:
My rule of thumb is:

* An error prevents a program from executing normally. It can no longer fulfill its purpose.
* A warning hints at an inconvenience that the program can deal with without tumbling. But the condition is worth reporting.

In a concrete case - like the `Connection` - the decision largely depends on the usage patterns. As an anecdote, many years ago, we implemented "clever" components that could adapt their behavior to their environment. For example, our version of L4Linux would always try to open a NIC session. If the connection could be established, L4Linux would initialize its virtual network card. If not, Linux would see no network device. Another example is init, which attempts to open a ROM session to obtain the dynamic linker at startup. If no dynamic linker is present, init would still run, but it would not be able to start dynamically linked binaries. With this "probing" approach, a failed `Connection` is often not an error but almost expected. So a warning would be more appropriate than an error.

In the meanwhile, our methodology of designing and implementing components shifted away from "clever" components to "stupid" but configurable components. Our modern components don't even try to adapt themselves to their environment. They obtain their configuration from a trusted party (their respective parents) and act according to the configuration. If the component fails to do so, this is an error. Coming back to the L4Linux example, today we would pass the information whether a NIC is present or not as configuration to L4Linux. If configured with NIC support, if would try to open a NIC session. If the session creation fails, it would immediately fail with an error instead trying to live without networking. So with our today's best practice of component design, a failed `Connection` is a fatal condition, which should be reflected by an error.

Until now, the error was simply an `abort` caused by an unhandled `Parent::Service_denied` or `Parent::Unavailable` exception, which is arguably difficult to interpret. With the error message added in my commit https://github.com/genodelabs/genode/commit/252350be628c19b032094ea65a2b7ea86fddd375, we will also see the session type and arguments.

trimpim, 2016-11-10:
With the patch from @nfeske the issue is solved for me.

closed 2016-11-10 by trimpim