genodelabs/genode #2168: intel_fb_drv: heap could not free memory block

rite, 2016-11-07:
Looks like a memory leak for me?

chelmuth, 2016-11-07:
This could also be double-free. Could you provide more context for the issue? After which runtime this message logged or does it appear sporadically?

rite, 2016-11-07:
The message is logged once, during startup of a fairly large scenario. Here's the filtered output:

```
09:54:52,566 [init -> intel_fb_drv] --- intel framebuffer driver ---
09:54:52,566 [init -> intel_fb_drv] INFO: i2c-core: driver [dummy] registered
09:54:53,706 [init -> platform_drv -> intel_fb_drv] assignment of 16 succeeded
09:54:53,706 [init -> platform_drv -> intel_fb_drv] assignment of 0 succeeded
09:54:53,706 [init -> intel_fb_drv] ioremap: mapped phys 0xe0000000 (size 16777216) to virt 0x8000000
09:54:53,707 [init -> platform_drv] Error: Device 0x0:0x0.0 is used by more than one driver - session 'intel_fb_drv'.
09:54:53,707 [init -> platform_drv] Error: Device 0x0:0x2.0 is used by more than one driver - session 'intel_fb_drv'.
09:54:53,722 [init -> platform_drv -> intel_fb_drv] assignment of 248 succeeded
09:54:53,722 [init -> intel_fb_drv] Warning: heap could not free memory block
09:54:53,722 [init -> intel_fb_drv] [drm:ERROR] blitter: timed out waiting for forcewake ack request.
```

commit 6c5422ae (2016-12-14): intel_fb: avoid heap warning

Fixes #2168
commit 3e0aeab8 (2016-12-23): intel_fb: avoid heap warning

Fixes #2168
closed 2016-12-23 by nfeske