genodelabs/genode #2170: os: timeout framework

m-stein, 2016-11-07:
The Alarm framework is currently used in two scenarios: In the timer driver to multiplex the hardware or kernel time-source amongst different timer sessions and in other components to multiplex a timer session amongst multiple timeout subjects. These use cases have common wrapper code for the alarm framework that may be moved to a new "timeout" framework that then may replace the alarm framework. Additionally we could add the feature of local time interpolation through timestamps.

m-stein, 2016-11-07:
My branch ``timeout`` shows how the new timeout framework is used in the timer as well as in a test component. If there are no objections, I would now start modifying other components that use the alarm framework to use the timeout framework instead.

nfeske, 2016-11-07:
I would change `Time_source` to be an abstract class. This way, the `Timeout_scheduler` does not need to be a template and could take its `Time_source` as reference argument to its constructor. This is beneficial for time sources that need constructor arguments. Strictly speaking, this would be even needed for the current `Time_source`, which needs access to the `Env` to create the timer connection. (your version still uses the deprecated argument-less `Timer::Connection` constructor.

Please look out for proper indentation/alignment. E.g.,`Timeout` constructor looks wrong.

m-stein, 2016-11-08:
@nfeske I've updated the branch according to your suggestions.

m-stein, 2016-11-10:
@nfeske The new version is now online.

m-stein, 2016-11-25:
@chelmuth I rebased it and also made the test ready for autopilot. Thanks for merging the RPC stuff :)

nfeske, 2016-11-30:
Now that the patches have reached the master branch, may we close this issue?

m-stein, 2016-11-30:
Of course :)

closed 2016-11-30 by m-stein