genodelabs/genode #2171: Make label prefixing more strict

nfeske, 2016-11-07:
While working on the async root/parent interfaces (#2120) and the accompanied adaptation of the component-creation procedure, I came across the following idea:

We used to pass a lot of arguments to the `Child` constructor such as the child's initial ROM/CPU/PD sessions, and the dataspaces for the ELF binary and the dynamic linker. With the new framework, it would be much more natural to let the child obtain those sessions using its regular session-routing mechanics that is subjected to the `Child_policy` anyway. This would not only make the child constructor much simpler, it would also ease the life of people who want to virtualize the services behind the basic sessions of the child (e.g., the loader, or noux).

If the child obtains its environment sessions using the regular session routing, the `Child_policy::resolve_session_request` method is called for these requests as well and we can conveniently route them to a corresponding service, most commonly a parent service. The problem is, however, that this policy method cannot distinguish the environment sessions from label-less sessions created by the child itself. If we want to create a child but don't want to allow the child to open a PD session by itself, the policy method cannot distinguish both cases because the current implementation of the `prefixed_label` function handles the special cases of the prefix of label being empty. In these cases, it sets the label to the non-empty part.

To resolve this ambiguity, I propose to always prefix the label with the " -> " characters between the prefix and the label even if one of both elements is not there. This way, the child's environment PD session would end up being labeled as "child_name" whereas a label-less PD session created by the child would become "child_name -> ".

I wonder, is there any good reason not to implement `prefixed_label` function as follows?

```
template <size_t N1, size_t N2>
inline Session_label prefixed_label(String<N1> const &prefix,
                                    String<N2> const &label)
{
  String<N1 + N2 + 4> const prefixed_label(prefix, " -> ", label);
  return Session_label(prefixed_label);
}
```

@ehmry Do you have an opinion about that?

nfeske, 2016-11-07:
While experimenting with this idea, I found two things worth noting.

* First, the LOG session is normally created as a Log_connection by the child. For this reason, all prefixes displayed by core's LOG service now end with a trailing " -> ", which is not nice. This problem can be fixed by opening the LOG session from the parent's context as an environment session.

* Second, label-based routing policies do not match if they referred to a unlabeled session. For example, in the demo.run script, the nitpicker policy contained the line `<policy label="status_bar" domain="panel"/>`. The status bar opens a label-less nitpicker session. Now, the policy would need to look like this `<policy label="status_bar -> ".../>`, or alternatively, `<policy label_prefix="status_bar"/>`.

In my opinion, both effects are quite ok. At the first sight, the second one looks a bit inconvenient but the " -> " separator makes it very obvious from where the session request originates, which I like.

Of course, the drawback is that the run scripts that use exact label matching must be revisited. But I think this is reasonable.

ehmry, 2016-11-08:
Simplifying the Child constructor would make things much easier for people trying to learn the framework, but I think a trailing ''->'' would be less intuitive for policy making. How is it that init can distinguish between an env request and a unlabeled request?

nfeske, 2016-11-08:
Note that the trailing " -> " must only be present when unlabeled connections are used. E.g., this is never the case for ROM sessions but it is common case for block sessions where a file system server typically requests only one block session and thereby opens an unlabeled block connection. I reviewed the run scripts and found that it won't be too invasive. In most cases where we use label-based routing/policy selection - like the <fs> nodes in the VFS - we actually use supply labels.

Regarding your question about init, in the current version, the env sessions are not requested via the `Child_policy::resolve_session_request` method. They are created outside the child and stuffed as arguments into the `Child` constructor. In the new version, the `Child` class consults the `Child_policy::resolve_session_request` method for both the initial (env) sessions and the sessions dynamically requested by the child. Here is where I encountered the ambiguity.

ehmry, 2016-11-08:
Ah, I understand, it is init-local resolution.

ehmry, 2016-11-08:
Ok, maybe its not as bad as I thought, and encouraging more precise labels is good.

On a related note, I wonder what you think of optionally not prefixing some requests? This would allow a sub-init configuration to be more easily modified without requiring a policy change in a parent. For example, I may want to add a framebuffer recording shim to a subsystem without worrying about a change in label causing a routing change below?

nfeske, 2016-11-09:
@ehmry I has a similar Idea: we could allow the label to be rewritten by the matching route. E.g.,

```
<start name="nitpicker">
  ...
  <config>
    ...
    <policy="cli_monitor -> untrusted" ... domain="default"/>
    <policy="cli_monitor -> trusted"   ... domain="panel"/>
  </config>
</start>

<start name="cli_monitor">
  ...
  <route>
    <service name="Nitpicker" label_prefix="login ->">
      <child name="nitpicker" label="trusted"/>
    <service name="Nitpicker">
      <child name="nitpicker" label="untrusted"/>
    <service/>
    ...
  </route>
</start>
```

This way, any child started by CLI monitor with the name "login*" will show up with the label "cli_monitor -> trusted" at nitpicker. All other CLI monitor children will appear as "cli_monitor -> untrusted". If one decides to change the name of the login program, only the `label_prefix` of the CLI-monitor route need to be changed.

The only concern that I have is that the meaning of the "label" attribute in the `<service>` nodes is very different from the meaning of the attribute in the `<child>` and `<parent>` nodes. The former is used as a selector for the policy, the latter is used to reassign a new label to the session. This might confuse users. However, I haven't managed to come up with a another concise name for the new label-assigment attribute. Or do I overthink it and naming the attribute "label" in both cases is just fine?

Emery, would that idea cover your use case well enough?

chelmuth, 2016-11-09:
I pretty much like your proposal for `<child ... label="adapted_label"/>` and for me it is not misleading at all. As I see it the routes are a table with selectors and the selected route defines to which component and with which arguments the request should be routed.

ehmry, 2016-11-09:
Yes, I like it, and I don't think it would be too confusing.

ehmry, 2016-11-09:
Its good too because it reinforces the idea that the significance of the label decreases from left to right.

nfeske, 2016-11-23:
While we are at it, I propose to make the session labeling mandatory. Right now, the labeling is applied by each `Child_policy` individually by overriding the `filter_session_args` method. By unconditionally applying the labeling, we can remove most overrides of `filter_session_args` (because most overrides apply the labeling but nothing else) and remove the `Init::Child_policy_enforce_labeling` class.

commit 3d66aa56 (2016-11-24): Make label prefixing more strict

This patch unconditionally applies the labeling of sessions and thereby
removes the most common use case of 'Child_policy::filter_session_args'.
Furthermore, the patch removes an ambiguity of the session labels of
sessions created by the parent of behalf of its child, e.g., the PD
session created as part of 'Child' now has the label "<child-name>"
whereas an unlabeled PD-session request originating from the child
has the label "<child-name> -> ". This way, the routing-policy of
'Child_policy::resolve_session_request' can differentiate both cases.

As a consequ [... cut]
commit c6c1eecc (2016-11-25): Make label prefixing more strict

This patch unconditionally applies the labeling of sessions and thereby
removes the most common use case of 'Child_policy::filter_session_args'.
Furthermore, the patch removes an ambiguity of the session labels of
sessions created by the parent of behalf of its child, e.g., the PD
session created as part of 'Child' now has the label "<child-name>"
whereas an unlabeled PD-session request originating from the child
has the label "<child-name> -> ". This way, the routing-policy of
'Child_policy::resolve_session_request' can differentiate both cases.

As a consequ [... cut]
closed 2016-11-30 by chelmuth