genodelabs/genode #2176: Genode::List::for_each

ehmry, 2016-11-18:
Eliminates writing some **for** loops that check linked list pointers.

nfeske, 2016-11-21:
I'd like to discourage the direct use of `List` in Genode components because it is quite error-prone:

* It is not thread-safe,
* It is easy to miss a remove operation, ending up with dangling pointers,
* Because `List::Element` is public, it is unclear who is the legitimate list head. For example, the original version of `Service` used to be a `List::Element` because services are often kept in a `Service_registry`. But the list-element part is not intrinsic to a `Service`.

Furthermore, the inherited `List::Element` objects pollute the class hierarchy. E.g., it is quite difficult to have one object present in multiple lists.

To sum it up, `List` invites problems. For this reason, I don't want to make it more attractive.

Instead, I try to come up with safer ideas for covering the use cases of lists. For example, a very common use case is remembering a bunch of objects so that we can destroy them at session-destruction time. To accommodate this use case, I just introduced a `Registry` data structure in commit  https://github.com/genodelabs/genode/commit/a5c0e8297623fd3caa96486ef17dff5548b9baad. It prevents several pitfalls of lists by design. Thanks to this data structure, I was able to untangle the `Service` from the list. Another recent data structure is the `Id_space`, which I introduced in commit https://github.com/genodelabs/genode/commit/2a4a68cb25ab6b790933bba09cbdb704f323f399. I introduced it as support for the asynchronous parent interface (and it helped me a lot detect programming and even design mistakes at compile time!). Now, I see plenty of places where the use of `Id_space` would make perfect sense (capability spaces, file handles, nitpicker view handles, object pool, just to name a few).

It is too early to tell whether those data structures succeed in eliminating most uses of lists or not. But I want to give it a try. For sure, we will still keep the list data structure (e.g., for the mechanics of `Registry`) but it will hopefully become a rare sight in component code.

ehmry, 2016-11-21:
I try to avoid list traversal but my use case at the moment is a push-down stack, but of the top of my head I cannot think of another use outside the VFS.

https://github.com/ehmry/genode/blob/pipe/repos/libports/src/lib/vfs/lwip/vfs.cc#L208
https://github.com/ehmry/genode/blob/pipe/repos/libports/src/lib/vfs/lwip/vfs.cc#L472

ehmry, 2017-01-23:
Closing in favor of ``Registry``.

closed 2017-01-23 by ehmry